pibooth/pibooth #8: Set option for one image instead of 4

werdeil, 2017-12-03:


werdeil, 2017-12-04:
I think that #15 shall be done before this one

anxuae, 2018-01-26:
The option in config already exists:
```
[PICTURE]
# How many pictures to take (4 max)
captures = 4
```
I think that the only thing to change is the `concatenate ` function.

werdeil, 2018-01-26:
This ticket is to add an option on the interface to change during the use (either a 1 shot or a 4 shot) not in the config

anxuae, 2018-01-26:
A yes, I've lost my mind

anxuae, 2018-01-27:
May be it can be done by using the "picture" button (and specific keyboard combo):
- long press
- several time press

werdeil, 2018-01-27:
Or a middle screen of 5 seconds where the suer choose between the 2

werdeil, 2018-01-28:
I have introduced the middle screen, if you want you can work on the action handling

anxuae, 2018-02-05:
https://github.com/werdeil/pibooth/commit/395d3fcbe41a673a193094ba9c51fcdf1351d3fa

closed 2018-02-05 by anxuae