pibooth/pibooth #17: Find a way to install the package when gphoto2 or cups is not installed

werdeil, 2018-01-28:
As it is very long and optional to have them it shall be possible to install pibooth via pip without them.

anxuae, 2018-01-29:
The only solution is to remove them from the`install_requires` setup. 

Raise an error if the user try to use them without installe them beforehand manually.

werdeil, 2018-01-30:
Then for me there are two options:
- Remove them from install_requires as you describe
or
- Add the workaround in the documentation.

Which one do you prefer?

anxuae, 2018-01-30:
the only possible workaround is `pip install pibooth --no-deps`, in that case all dependencies have to be installed manually.

anxuae, 2018-02-01:
Ok for workaround or I remove the dependencies

werdeil, 2018-02-04:
workaround it is

anxuae, 2018-02-05:
https://github.com/werdeil/pibooth/commit/1ee5b8569bab82e712ff3043fac513e19d03f64d

closed 2018-02-05 by anxuae