pibooth/pibooth #25: GPIO for LED-Lighting

thifi, 2018-03-14:
Dear werdeil,

sorry for my bad english and my questions, I'm electronic engineer in hardware, not a software developer.

First many thanks for sharing your code for the pibooth, I want to build one for an event of my 14years old daughter called "Jugendweihe" in Germany in may.

I used your code and after some hours of study it worked on my Pi3, but I have 2 questions about:

1. Is it possible to implement the control of another GPIO of the Pi to turn on/off an LED-light for the picam or an ambi-light on the box?

2. I get after printing on a Selphy 1300 a postcard in portrait-mode, is it also possible to design the pictures in landscape-mode?

Many many thanks for your understanding of my problems and you help.

Best regards
Thilo

werdeil, 2018-03-15:
Hi Thilo,

Can you tell us more on this third GPIO control? For which purpose? When do you want it to be activated in the current sequence of events?

For the second topic, I have opened a second issue #26, can you add more information on it on what you want. As we defined the whole interface for portrait mode photos we need to clearly understand your need before trying any change for it.

thifi, 2018-03-15:
Dear werdeil, many thanks for your question.
I want to switch on a LED photo lamp for better pictures with the picam.
It would be the best to switch it on when the live preview is start to take pictures and switch it off if all pictures taken. 
Another optional GPIO could used for animation light (LED strips) and activated when program starts.

werdeil, 2018-03-17:
If you want a lamp for better pictures why not having it always on? I just want to be sure that we don't add a GPIO output for nothing.

thifi, 2018-03-17:
I want to use some COB-LEDs for lighting, If they alway on the LED produce to much temperature in the box, so it would be better to turn it on with the live view and off after all 4 pictures taken.
The optional GPIO for ambi-light could activate with the program start.

anxuae, 2018-03-17:
Hi Thilo,

Here is the proposal with 2 supplementary LED:

- Physical pin 29 : for ambient-light, set to 5V at `pibooth` startup, set to 0V at exit
- Physical pin 31 : for COB-LED, set to 5V at preview startup, set to 0V at preview end

I'm not expert in electronic, but in think that Raspberry Pi will not be enough powerful for consuming lights. Thus decoupling with external power source may need necessary.

Does it corresponds to your needs?

[image]

thifi, 2018-03-17:
Hi anxuae,
that sounds great, many thanks to you.
I will use a NPN-MOSFET as an amplifier to switch the LED-lights.
Many thanks for your attention.
How should the software be updated?

thifi, 2018-03-17:
Hi anxuae,
I make pip uninstall pibooth and pip install pibooth, but there is the same version as before.
Can you help?
Best regards

werdeil, 2018-03-17:
We need to release a new pip version, we will try to do it before the end of the week-end.

While you wait can you provide us a translation of the wording in german? It will permit us to have a third supported language, here are the wordings:
Choose your layout
"Let's go !"
"Thanks"
"Take a photo"
"You can still print this photo"
"Print the photo?"
"Processing"

thifi, 2018-03-17:
Here it is:
Choose your layout  -- Wähle dein Layout
"Let's go !"  --  Los gehts!
"Thanks"  --  Danke
"Take a photo"   --  Foto aufnehmen
"You can still print this photo"  --  Sie können dieses Foto immer noch ausdrucken
"Print the photo?"  --  Foto drucken?
"Processing"  --  Bearbeitung
If you need other informations, please contact me.

anxuae, 2018-03-18:
Features available in release https://github.com/werdeil/pibooth/releases/tag/0.0.7
Package uploaded on Pypi

closed 2018-03-18 by anxuae