pibooth/pibooth #26: Landscape mode

werdeil, 2018-03-15:
Following second topic on issue #25, we need to implement a landscape mode

thifi, 2018-03-15:
Thank you for opening this issue.
Actual the software takes 4 pictures in portrait mode and combine it to a postcard in portrait mode with footer 1 and 2 on the small side.
I wish a postcard in landscape mode with 4 pictures on it and a footer and a header on the large sides.
Another option is to use a personal template as backround like attached.

[image]
[image]
[image]

anxuae, 2018-03-18:
Features available in release https://github.com/werdeil/pibooth/releases/tag/0.0.7
Package uploaded on Pypi

The issue on template as background is not yet addressed (open a new ticket?)

werdeil, 2018-03-18:
Closing issue

closed 2018-03-18 by werdeil