pibooth/pibooth #29: Error in 0.0.7 , doesn't start

thifi, 2018-03-18:
Hi,
I uninstalled, rebooted and installed pibooth in version 0.0.7 and started pibooth normally:

pi@raspberrypi:~ $ pibooth
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 7, in <module>
    from pibooth.ptb import main
  File "/usr/local/lib/python2.7/dist-packages/pibooth/ptb.py", line 19, in <module>
    from pibooth.view import PtbWindow
  File "/usr/local/lib/python2.7/dist-packages/pibooth/view.py", line 6, in <module>
    from pibooth import pictures
  File "/usr/local/lib/python2.7/dist-packages/pibooth/pictures/__init__.py", line 6, in <module>
    from pibooth.config import PtbConfigParser
  File "/usr/local/lib/python2.7/dist-packages/pibooth/config.py", line 30
    "footer_text1": ("Footer 1", "First text displayed"),
                 ^
SyntaxError: invalid syntax
pi@raspberrypi:~ $ 

This was the reaction. Can you help me?

anxuae, 2018-03-18:
I've uploaded v0.0.8 on Pypi, can you reinstall and try again?

Thanks

thifi, 2018-03-18:
Version 0.0.8 starts, but there are some probs too:
In layout I could not change with the the Keyboard (P or CTRL-E) between 1 or 4 pictures.
The Live-Preview is in portrait-mode, I changed it to landscape in config but live is in portrait again.
The orientation of the final image is in landscape mode, but the pictures are in portrait mode (very small)
There are no Footer or Header Text in the final image.

The GPIOs working perfect, thank you very much.
Thilo

anxuae, 2018-03-18:
Hi Thilo,

- To change between 1 or 4 photo, you have to use LEFT or RIGHT arrow keys (see the README https://github.com/werdeil/pibooth#commands)
- There are 2 way to manage the final rendering (which may not be obvious):
   - the resolution ( in `[CAMERA]` section of the config) which is the format of the captured photo. Now, the default  is `portrait`. You have to change in order to the photo has a width larger that the height (e.g. `(2464, 1934)`  ).
   - the orientation ( in `[PICTURE]` section of the config) which is the format of the final picture after concatenate all picture. Put the `landscape` value.

werdeil, 2018-03-18:
For the resolution I would advise something like 3280x1934, it looked quite good on my side.

For the footer, what did you write in the config for the following keys: footer_text1, footer_text2? If the key are empty no footer will appear.

thifi, 2018-03-18:
I tested it with 4 Photos in 16:9 format and now it will be ok, but the footer1 and 2 are absent.
footer_text1 and footer_text2 are TEST1 and Test2.
[image]

werdeil, 2018-03-18:
Then is your text_color different from your background color?

thifi, 2018-03-18:
Here is my config:

thifi, 2018-03-18:
I think there is a big problem.
I reinstalled the pibooth and deleted the pibooth.cfg in /home/pi/.config.
After that the program installed an old pibooth.cfg without the configuration portrait/landscape-mode and I get no Footer also in portrait mode.
Is there a cache or so? How can I delete all of pibooth to reinstall complete?
Thanks for your help.

anxuae, 2018-03-18:
Be sure that you have no clone of the project on your RPI.
Got to the following paths and remove any folder, files containing `pibooth`:
- usr/local/lib/python2.7/dist-packages
- usr/local/lib/python3.5/dist-packages
Reinstall using `sudo pip install pibooth`
Reset the configuration using `pibooth --reset`

thifi, 2018-03-18:
I removed Pibooth under python2.7, there are no packages under python3.5.
Then I reinstalled Pibooth with PIP:

pi@raspberrypi:~ $ sudo pip install pibooth
Collecting pibooth
  Using cached pibooth-0.0.8-py2.py3-none-any.whl
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (from pibooth)
Requirement already satisfied: Pillow in /usr/local/lib/python2.7/dist-packages (from pibooth)
Requirement already satisfied: gphoto2 in /usr/local/lib/python2.7/dist-packages (from pibooth)
Requirement already satisfied: Pygame in /usr/lib/python2.7/dist-packages (from pibooth)
Requirement already satisfied: pycups in /usr/local/lib/python2.7/dist-packages (from pibooth)
Requirement already satisfied: picamera in /usr/lib/python2.7/dist-packages (from pibooth)
Installing collected packages: pibooth
Successfully installed pibooth-0.0.8
pi@raspberrypi:~ $ 

Reset Configuration and started Pibooth, no Footer ......
Attached the plain config-file.
[pibooth.txt](https://github.com/werdeil/pibooth/files/1823187/pibooth.txt)

I don't know, what is wrong.....
Is this ok:
 Using cached pibooth-0.0.8-py2.py3-none-any.whl
?

anxuae, 2018-03-18:
The cache used by Pypi is only to avoid download again the package, there is no impact

anxuae, 2018-03-18:
Ok, I can reproduce the issue, I need to investigate

thifi, 2018-03-18:
I sent you the config after pibooth --reset as a txt-file in the last post.
I have a copy of the old config.

anxuae, 2018-03-19:
I found the reason, and solution but need a new release.
I will fix it this evening at home

anxuae, 2018-03-20:
fixed in https://github.com/werdeil/pibooth/commit/21cb7c4d1794994f31a0286b23b8f40a680c49c3

anxuae, 2018-03-20:
Available in v0.0.9

thifi, 2018-03-20:
It works, the footers also ok.
Thank you.

closed 2018-03-20 by anxuae