pibooth/pibooth #30: Abortion when no printer installed

Dommsel, 2018-03-20:
After taking fotos and displaying them, the programm aborts when no further button is pressed. I've installed pibooth with no-deps, because I want to conduct it without printing photos.

Is it possible to run the programm without the printing option? My idea is, after diplaying the taken photos, return to start for a new session without fading in the printing option.

anxuae, 2018-03-20:
Hi Dommsel,

Yes it is normally possible to run `pibooth` without printer. Can you publish here the traceback error?

Thanks

Dommsel, 2018-03-20:
Here you are:

`pi@raspberrypi:~ $ pibooth
[ INFO    ] pibooth           : Starting the Photo Booth application...
[ INFO    ] pibooth           : Show picture choice (no default set)
[ INFO    ] pibooth           : Set 1 picture(s) mode
[ INFO    ] pibooth           : Start new pictures sequence
[ INFO    ] pibooth           : Take picture and save it in home/pi/usbstick/pibooth/2018-03-19-21-59-54/ptb000.jpg
[ INFO    ] pibooth           : Creating merged picture
[ INFO    ] pibooth           : Save the merged picture in home/pi/usbstick/pibooth/2018-03-19-21-59-54/2018-03-19-22-00-02_ptb.jpg
[ INFO    ] pibooth           : Display the merged picture
[ INFO    ] pibooth           : Send final picture to printer
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/dist-packages/pibooth/ptb.py", line 415, in main
    app.main_loop()
  File "/usr/local/lib/python2.7/dist-packages/pibooth/ptb.py", line 370, in main_loop
    self.state_machine.process(events)
  File "/usr/local/lib/python2.7/dist-packages/pibooth/states.py", line 57, in process
    self.active_state.do_actions(events)
  File "/usr/local/lib/python2.7/dist-packages/pibooth/ptb.py", line 43, in do_actions
    self.app.printer.print_file(self.app.previous_picture_file)
  File "/usr/local/lib/python2.7/dist-packages/pibooth/controls/printer.py", line 152, in print_file
    raise EnvironmentError("No printer found (check config file or CUPS config)")EnvironmentError: No printer found (check config file or CUPS config)`

Dommsel, 2018-03-20:
Problem occures when the button 2 is pushed by displaying the image. Should be possible to hide the "Print foto afterwards"-function or disable button 2 at this time

anxuae, 2018-03-20:
Yes you can hide the print state by setting the `print_delay=0` (see configuration details: https://github.com/werdeil/pibooth#configuration )

anxuae, 2018-03-20:
Does it correspond to your needs? Can I close the issue?

Dommsel, 2018-03-24:
Hi anxuae, sorry for my delayed answer!
The suggested setting `printer_delay=0` does not help me so much. My problem is marked in the upload. By hitting `button 2` at this point, the program will abort. So I don't want do give users the possibility to print.
Don't worry about the monitor, it's 4:3 resolution ;)
[image]

anxuae, 2018-03-26:
Ok, I will look for a fix. The new behavior will be to skip print actions and print message if:

- `pycups` is not installed
- no printer configured in `CUPS`

Dommsel, 2018-04-03:
Hey anxuae,
thanks for your help, but I've got a last, dumb question :(
I justed started working with my raspi and so I still got not the possibility to gain much experience. I want to have commit #227fc3f run, but I don't know how. My steps were:
- uninstalling pibooth by using `pip uninstall pibooth`
- downloading zip from #227fc3f repository
- unzip data to python directory in usr/local/lib

By starting pibooth through terminal I got the message: `bash: pibooth: Command not found`. Can you give me a short manual how to get the program run?

anxuae, 2018-04-04:
Hi Dommsel,

The commande `pibooth` is only available for official release on the [pypi](https://pypi.python.org/pypi) server.

If you want to use an unstable version, follow this steps:

- clone from github: `git clone https://github.com/werdeil/pibooth.git`
- in a terminal, got to the cloned directory: `cd pibooth`
- run the command: `PYTHONPATH=. python3 pibooth/ptb.py`

Be careful, sometime we push commits which are not fully debugged, thus it may not work.

anxuae, 2018-04-04:
Available in [v1.0.0](https://github.com/werdeil/pibooth/releases/tag/1.0.0)

closed 2018-04-05 by anxuae