pibooth/pibooth #31: Background picture

thifi, 2018-03-20:
Is there optionally the possibility to change the background RGB-Color to a uploaded .jpg or .png file to make the printed postcard more interesting?

anxuae, 2018-03-20:
Implemented in https://github.com/werdeil/pibooth/commit/0ad78a3ead20d472a154c74aeb159df9462d8e00

anxuae, 2018-03-20:
Available in v0.0.9

thifi, 2018-03-20:
It works like a charm. Thank you.

Is there a possibility to set a final image with 4 pictures as default?
In the log: 
(INFO)  pibooth: Show picture choice (no default set)

werdeil, 2018-03-21:
Just to understand, you want to avoid the selection between 1 and 4 photos? Or that if nothing is selected the 4 photos are chosen?

For the second part, we had it at first but the tests we made with users showed that it wasn't understandable. That's why we changed it to go back to the main menu if nothing is selected.

thifi, 2018-03-21:
I want to make a postcard with 4 photos with a special backgraund for these 4 photos, so it is important, that the user select these option. 
One way is to make the 4 photo selection as default (and if nothing selected, the 4 photos where taken). The other way is to make the selection in the config without a choise for the user (avoid the selection in frontend).

thifi, 2018-03-21:
Many thanks again for the option background....

closed 2018-03-21 by werdeil