pibooth/pibooth #32: Crash in hybrid mode

werdeil, 2018-03-20:
Here is the traceback, seems to happen in very low luminosity.

Take picture and save it in /home/pi/Pictures/pibooth/2018-03-20-19-07-45/ptb000.jpg
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 11, in <module>
    load_entry_point('pibooth==0.0.8', 'console_scripts', 'pibooth')()
  File "/usr/local/lib/python3.5/dist-packages/pibooth-0.0.8-py3.5.egg/pibooth/ptb.py", line 418, in main
    app.main_loop()
  File "/usr/local/lib/python3.5/dist-packages/pibooth-0.0.8-py3.5.egg/pibooth/ptb.py", line 373, in main_loop
    self.state_machine.process(events)
  File "/usr/local/lib/python3.5/dist-packages/pibooth-0.0.8-py3.5.egg/pibooth/states.py", line 57, in process
    self.active_state.do_actions(events)
  File "/usr/local/lib/python3.5/dist-packages/pibooth-0.0.8-py3.5.egg/pibooth/ptb.py", line 146, in do_actions
    self.app.camera.capture(image_file_name)
  File "/usr/local/lib/python3.5/dist-packages/pibooth-0.0.8-py3.5.egg/pibooth/controls/camera.py", line 322, in capture
    file_path = self._gp_cam.capture(gp.GP_CAPTURE_IMAGE)
gphoto2.GPhoto2Error: [-110] I/O in progress

werdeil, 2018-03-21:
As the issue comes from gphoto2 we can only catch the exception. but what shall we do in this case?

The option I think of:
- Just catch the exception, as the number of captures is not incremented the process will retry. But there is a huge risk of capture loop.
- Catch the exception, show a "Oops there seems to be an issue screen" (to be designed) and go back to the start screen.
- Leave the crash as it is.

@anxuae what do you think?

anxuae, 2018-03-21:
Second option seems good, this can be directly implemented in the state machine
 as a global mechanism in case of exception (keep traceback displayed if verbose option for debugging purpose)

anxuae, 2018-03-22:
To be tested https://github.com/werdeil/pibooth/commit/7e64847f13ff749995c2fdd3541830de2b8925d7

anxuae, 2018-03-28:
Can this issue be closed?

werdeil, 2018-04-01:
Following commit  3599971 this issue can be closed.

closed 2018-04-01 by werdeil