pibooth/pibooth #33: Add an option in the config to have only one choice

werdeil, 2018-03-21:
Request from @thifi in #31

"""
I want to make a postcard with 4 photos with a special backgraund for these 4 photos, so it is important, that the user select these option.
One way is to make the 4 photo selection as default (and if nothing selected, the 4 photos where taken). The other way is to make the selection in the config without a choise for the user (avoid the selection in frontend).
"""

My choice would be to do the second option, hence the title of the issue.

anxuae, 2018-03-21:
Ok for 2nd option. The first one have the drawback of the timing window to select the photo, which may be limiting if you are far of the button or inattentive.

anxuae, 2018-03-25:
Implemented in https://github.com/werdeil/pibooth/commit/6a43a427341a678f2a0b7534926fb51eebe1fa7c

anxuae, 2018-04-04:
Available in [v1.0.0](https://github.com/werdeil/pibooth/releases/tag/1.0.0)

closed 2018-04-05 by anxuae