pibooth/pibooth #35: Use screen as flash

bderenzo, 2018-04-20:
Hi,

Is it possible to add a blank overlay on the screen just before taking picture (on whole screen, even on camera preview) ? In order to use the screen as a flash.

thanks in advance,
bderenzo

anxuae, 2018-04-25:
Hi bderenzo,

I add the option `preview_stop_on_capture` to stop the preview during the picture capture, so the flash should be visible on the whole screen.

However, I have already done some tests, the light given by the screen is really to low and not efficient for the pictures. I really suggest you to use an external flash/light.

anxuae, 2018-05-01:
Hi,

Have you try the modification?
I will generate a new version today, can I close the issue?

Thanks

ludodarius, 2018-05-01:
Bonjour,

J'ai installé pibooth il y a un mois et j'en suis super content. 

Je vais l'utiliser pour mon mariage début juin.

J'ai fait sudo pip3 install pibooth lors de l'installation.

Pouvez - vous m'aider en me donnant la procédure pour mettre à jour pibooth en fonction des évolutions sans devoir refaire l'installation ?

Merci.
Ludovic

anxuae, 2018-05-01:
Bonjour Ludovic,

Content que ce logiciel te plaise, n'hésites pas a nous faire des retours.

Concernant la mise a jour, il te suffit juste de relancer la commande suivante à chaque fois que nous publions une nouvelle version (celle précédemment installée sera écrasée "proprement" par la nouvelle) :

`sudo pip3 install pibooth`

Tony

closed 2018-05-02 by anxuae

ludodarius, 2018-05-04:
Merci Tony.

J'enverrai des photos du final en juin.

Quand je lance la commande sudo pip3 install pibooth j'ai le retour
requierement already satisfied et lorsque je regarde la version je suis
toujours en 1.0

A priori il n'y a pas d'écrasement de la version installée sur le raspberry
:(

Comme je compte me servir également du pibooth pour mon boulot (je suis
bibliothécaire et organise régulièrement des manifestations) l'idée est de
profiter des différentes améliorations sans refaire l'installation.

Je suis preneur d'une autre idée et par défaut j'essaierai de modifier les
droits administrateur avec un chmod -R 777 en superutilisateur pour voir
(je pense que cela n'aura aucun effet puisque je n'ai pas un message avec
accès refusé mais juste que le contenu existe déjà).

Continuez votre super boulot.

Merci pour l'aide.
Ludovic

bderenzo, 2018-05-04:
Bonjour Ludovic,

Ajoute simplement l'option -U dans ta commande ça devrait permettre la mise
à jour.

Baptiste.

ludodarius, 2018-05-04:
> t66syZcE8Z7KNwks5tuNVegaJpZM4TeET0

Merci Baptiste.

J'essaie ce midi de faire :

sudo pip3 install -U pibooth

Je ferai un retour pour dire si cela a fonctionné.

Ludovic