pibooth/pibooth #36: Touchscreen Support

bdubs85, 2018-05-23:
Is it possible to have touchscreen layout/support rather than a button configuration?

werdeil, 2018-05-26:
We can try to add a support to clicks, it may work on your touchscreen but as we don't have one we won't be able to test it.

anxuae, 2018-06-01:
Hi bdubs85,

Have you got a RPI touchscreen to tell us if the screen acts as a mouse?

thanks

bdubs85, 2018-06-01:
I actually had an elo touchscreen.

werdeil, 2018-06-05:
Ok I'll work on making parts of the screens clickable, I'll keep you posted when I have something testable.

werdeil, 2018-06-25:
@bdubs85 can you try the first dev from @anxuae , I'v tried it on my pibooth and it seems to work well with a mouse, I can click on the 2 parts of the screen, seems a good start.

bdubs85, 2018-06-25:
I just moved so my booth is packed up at the moment, I will check when I
get it unpacked.

nicolasborn, 2018-07-25:
@werdeil: do you have a video of your photobooth in action?

werdeil, 2018-08-05:
@nicolasborn I have uploaded a [video](
https://youtu.be/yFHqDGvck6Q) of my pibooth in action, unfortunately my picamera is down so I only have the mode with a DSLR.

nicolasborn, 2018-08-06:
hi @werdeil :+1: 
thanks for the video. unfortunately i cannot watch it. is the link correct? please also check the privacy settings of your video, maybe it is private.

werdeil, 2018-08-06:
I have changed the privacy settings, it shall now work.

nicolasborn, 2018-08-12:
thank you, it worked. nice photobooth! i've already built a photobooth a few years ago with a raspberry pi 1 and a touch screen but the case isn't very nice and the code could be better :-) and i always had problems with the light. maybe i'll build one with your software during winter.

anxuae, 2018-09-03:
I think this issue can be closed, the topic is now carried out by #41