pibooth/pibooth #37: Question:  Can we print a different picture than the one showing at the print page

Aurel-Charles, 2018-06-04:
Hi guys!
here is my thing:
Ive got Rpi + Canon selphy 1300 printer.
I'v modified the "concatenate" so i can print 2 same strips of 4 pics side by side. (in 4 pics mode of course). Then people just have to cut it with scissors and keep one for themself and leave one for the married 
But i want to show only one strip on the print screen.

Is there a way to merged one picture that is shown in the application at the print screen and then at the beginning afterward, and merged another merged picture (from the same photos taken) that will be sent to the printer (in print button is pressed)

is that clear?????? can you help?

i'm new in all of this but i find it fun!!

Thanks anyway, Pibooth application is so good. Great job!

werdeil, 2018-06-05:
Hi,

I like your idea of the two same strips.
I understand you need but for me I would have done it the other way:
- create a picture of one strip
- change the way the print is done to send 2 strips in one image to the printer
This way we don't have the need to change the showing process. What do you think?

Aurel-Charles, 2018-06-05:
hey,

yeah I agree with you and think its a good idee. I thought that could work, I can't figured out how to send 2 strips to the printer.
So if you can help with that that would be nice.

anxuae, 2018-06-06:
Hi,

Maybe we can have a hope using the CUPS options: `{"copies": 2, "number-up": 2}` when the picture is sent to the printer (means 2 copies and 2 pages per sheet).

If it works, an option on printer may be considered in the config.

Aurel-Charles, 2018-06-11:
Hi
I don't know how to do this, what am I suposed to type and where ? 
in the console I've tried :  lp -o collate=true -2 
but it just stretch the strip on the width and printer was stuck to print the second picture

anxuae, 2018-06-11:
You should try something like that:
``
lp -n 2 -o number-up=2  2018-06-05-23-15-53_ptb.jpg
``

(the dictionary from my previous post is the way to pass options to the `pycups` lib)

Aurel-Charles, 2018-06-12:
It doesnt work, just tried it but it just print a stretched picture twice

anxuae, 2018-06-16:
After long time spent in the CUPS manual and options, I think that this is not so obvious to manage this feature using the printing options.

anxuae, 2018-06-16:
Hi,

I pushed an patch which permits to choose the number of picture copies to be printed on one page (use the `nbr_copies` option). However the final rendering is really specific to the users need and it seems to be difficult to make it enough generic for the `pibooth` application.

ax

werdeil, 2018-06-19:
@Aurel-Charles can you send us your modified version of the concatenate file so we can try to integrate it somehow? You can attach the file to your answer.

Aurel-Charles, 2018-06-19:
Hey sorry i was away.
So here is the code, so i can get 2 strip of four pics in landscape mode . probably not conventional btu i got what I want . (also change the font)

[concatenate_printer.txt](https://github.com/werdeil/pibooth/files/2117263/concatenate_printer.txt)

anxuae, 2018-09-03:
Hi Aurel-Charles,

What's new about this topic??

anxuae

Aurel-Charles, 2018-09-13:
Hi .... sorry for late reply....
So I havent work much on that issue unfortunately. My neighbour might have look  see if he can help....

And by the way , except that screen showing 2 photos strips (in 4 pics mode) , my pibooth run well!! 
[image]
(just added some leds on sides and ring led)
 people at the wedding went crazy all night long !! **thanx a lot for you work and help**!!

jasharrar, 2018-09-14:
That looks great, what sort of printer are you using. I'm using a selphy. But trying to find one that hold more paper.

anxuae, 2018-09-14:
Hi Aurel-Charles,

Thanks for your feedback. Nice photobooth, I hope that the wedding was magic.

@jasharrar, we have used a common printer which accept the photo paper

anxuae

Aurel-Charles, 2018-09-17:
@jasharrar , Im also using a selphy + cups but there is some troubles, like sometimes, for no reason, the printer just say in communication with computer but doesnt print (it happened quiet a lot).  so i'm looking also for an other printer ... one with more paper and more efficient!

Aurel-Charles, 2018-09-23:
yep you can!

closed 2018-09-23 by Aurel-Charles