pibooth/pibooth #39: Neopixel LED ?!

Aurel-Charles, 2018-06-14:
Hey guys ! 

I've been trying the Neopixel LED strip , ring ... few days ago . Do you think that there is a way to integrate a script to the Pibooth so it get madly colorfull.
I've been trying but no result for now.
I've just followed these link for the setup : 
https://learn.adafruit.com/neopixels-on-raspberry-pi/overview
https://github.com/jgarff/rpi_ws281x

Its just an Idea i was thinking about just make it run aside so the ambient LED Strip color is changing a bit for each picture . I also thought putting a Led ring around the piCamera so if i could synchronise it for the start of the capture that would be amazing. 

anyway here is my idea , I know it a bit specific but if you guys can help in anyway that would be great!!

A

werdeil, 2018-06-19:
It can indeed be interesting to integrate some interface to neopixel strip/ring but I don't think we should integrated the full neopixel experience in the pibooth project. 

We already included 2 GPIO as an interface for the start preview and startup. Maybe we need to have a more generic interface.

Do you have any specific application that you can think of with a neopixel ring? I have a ring on my side but I have not found a use for it.

Aurel-Charles, 2018-06-19:
Hey so after few test , I manage to make it run so the my strip led is changing color randomly each time a picture is taken , and make the led of the ring turn on for the capture event . also a not conventional code as I just insert few lines here and there,  but works for me. 
My led strips are going to be on each side of my screen as a ambiant light, and the ring around the picamera.

anxuae, 2018-09-23:
I close this issue as it was solved by a specific développent

closed 2018-09-23 by anxuae