pibooth/pibooth #41: full screen touchscreen

jasharrar, 2018-09-03:
hi,  
firstly, this is an excellent project.  im really impressed, it looks great and its fairly easy to use for someone like me!
i have tried using the touch screen events (take pic etc) which work very well however when i use it as full screen (via the config) the touch screen doesnt work.  is there a work around for this?

Also for the GPIO buttons i am having some issues, it either doesnt work or if it does it will read multiple presses.  Just for clarification the definitions in the config are they for the physical pins or the GPIO numbers.  

many thanks in advance.
Jas

(currently using 1.02 and a raspberry pi official 7 inch screen)

anxuae, 2018-09-03:
Hi jasharrar,

Thank a lot for your feedback (if you appreciate this soft, you can also add a star :blush: ) !
I think this is because the mouse is disabled during the full screen mode, this could be fixed by setting an invisible cursor instead (https://stackoverflow.com/questions/40627934/use-pygame-mouse-while-hidden).

About the GPIO pins: the physical pins numbering is used (which is more obvious that the BCM numbering), see: https://pinout.xyz/# .

anxuae

jasharrar, 2018-09-03:
Thank you, that's very much appreciated. Also if you could let me know how to apply the fix that would be great (I'm pretty new at this).

Have a great evening

Jas

anxuae, 2018-09-03:
Please have try using the developing instructions here: 
https://github.com/werdeil/pibooth#install-developing-version

jasharrar, 2018-09-03:
think i have done something wrong, i followed the dev instructions and when i try to run pibooth i get sudo: booth: comand not found.  What am i doing wrong?
Or is there a way to edit the code to get the mouse working?
thanks again

Jas

anxuae, 2018-09-04:
Sorry a last instruction is missing: you add to declare the path to the executable. Before running the command, tip:

`export PATH=$PATH:$HOME/.local/bin`

You can also add the line to the `~/.bashrc` file to make the export permanent.

anxuae, 2018-09-04:
Another solution :  remove the `--user` option. in the cloned repository, tip:

`pip3 install -e .`

jasharrar, 2018-09-04:
i struggled with the above, however i copied the text from the text from the commit into the relevant files and it has worked perfectly  Thank you for all of your help.
You are awesome!
Jas

closed 2018-09-05 by anxuae