pibooth/pibooth #42: Tell us! What feature is missing

anxuae, 2018-09-03:
This is not an issue so much as a lightweight way of gathering information about which feature is missing to pibooth. This is mostly to satisfy our curiosity, but might also help us decide how to evolve the project.

So, if you use pibooth, please post here and tell us more!

jasharrar, 2018-09-11:
Not an issue but this is the booth that I made with your help.
https://www.instagram.com/p/BnmO5Lmni1n/?utm_source=ig_share_sheet&igshid=6rxp83eq3j47

anxuae, 2018-09-14:
Hi Jasharrar,

Thanks a lot for your feedback. It's look really great!

anxuae

Aurel-Charles, 2018-09-17:
Hi guiys! 
I was wondering if.... maybe we can have all the final photos (after "processing")  in a folder ... and all the picture taken in an other one . don't know if we can manage to have the date+time on each file name so there all unique (meaning avoiding "ptb000,ptb0001.....)

Cause when I uploaded on a google photo or whats so ever ... its a bit of a mess or really long to order evrething!.
It might be easier to manage photos after a party.

Its just an idee but let me know what do you think about it!

Aurel

werdeil, 2018-09-17:
Hi Aurel-Charles, it is a nice improvement indeed, I have opened a new issue for that #43. I'll try to work on this soon.

anxuae, 2018-09-23:
Hi deha89,

The difficulty depends on how you want to change the image effect: if the effect is permanent, we can easily manage it in the config file. Else if the user have to chose the effect, it will be more complicated. Can you tell me more about your need?

deha89, 2018-09-23:
Hey anxuae, thanks for the fast reply. It would be fine for me to set the effect permanent. Via config file sounds good. Shortly I also thought about it but i didn't find the line, where the access on the picamera api happens. Could you help me?

anxuae, 2018-09-23:
Ok I created thé #45 to manage this change request.

anxuae, 2018-09-24:
Hi @jasharrar and @Aurel-Charles,

I allowed myself to use your photos and videos in order to make a short demo gallery of what is it possible to do with the pibooth: https://github.com/werdeil/pibooth/wiki .
Anyway if you are disagree, tell me and I will remove them.

anxuae

theonlyuser, 2018-09-25:
Hey,

I just found this nice Project, thanks for the huge work you did!

I missing the possibility to disable the live-view. My Sony Alpha A6000 isn t able to have live-view / capture view. Is it maybe possible to deaktivate the live-view, because otherwise it leads to a problem, and no picture is taken.

Thanks in advance

regards

werdeil, 2018-09-26:
Hi @theonlyuser Can you open a new issue explaining the issue? is it that the transition between live-view and capture is not working or that the A6000 doesn't provide the preview? I'll try to find a Sony DSLR to test on my side.

Wikinaut, 2019-02-18:
### Suggestion: add Webcam support
I would like to be able to use PiBooth also with a standard **Webcam** like Logitech C922 PRO.

moskate49, 2019-03-01:
suggestions:
-sauvegarde des photos sur répertoires multiples (SD et copie sur clée USB par exemple)

iditude, 2019-07-06:
> suggestions:
> -sauvegarde des photos sur répertoires multiples (SD et copie sur clée USB par exemple)
> -un bouton pour eteindre proprement le raspberry sans clavier

Hello,

For the one button to shutdown raspberry, this already exists. More info here. Maybe it could be part of doc, but this can work without pibooth. https://forums.framboise314.fr/viewtopic.php?t=4423

Simple as adding dtoverlay=gpio-shutdown in file /boot/config.txt and plugging a pushbutton to pin 5 and 6.

G

Bonus: It also starts the pi

moskate49, 2019-07-06:
Works great thx
What about multiple folders save??
A play mode for looking all pictures will be a greath feature for review photo on another day

thifi, 2019-07-09:
I would also like to say thank you for your great work and the support I got during the construction of the pibooth photo-box.
Here is a link to some pictures of my pibooth photo-box.
[https://www.dropbox.com/sh/3suovagm3uizh90/AACXiAlKOYQA-NPyAQo7YzhWa?dl=0]([https://www.dropbox.com/sh/3suovagm3uizh90/AACXiAlKOYQA-NPyAQo7YzhWa?dl=0)
Have a great day...
Thifi

anxuae, 2019-07-10:
Hi Thifi,

Thanks a lot for your feedback, I attached your photo-box pictures to the pibooth'wiki.

regards,
anxuae

anxuae, 2019-07-10:
Hi moskate49,

> What about multiple folders save??

I'm not sure to understand the need? The ``save``directory can be defined in the config: a path on the SD card or a path to an USB stick. Why do you need to duplicate your picture in two different places?Never the less, you can open a specific issue for this topic

> A play mode for looking all pictures will be a greath feature for review photo on another day

Think this can be useful, but difficult to integrate in ``pibooth``. There are lots of application that already do this, for instance: https://pimylifeup.com/raspberry-pi-photo-frame/ .

regards
anxuae

moskate49, 2019-07-15:
i want save pictures in 2 folder 1 on raspberry inside the photobooth and a copy on usb key outside of the photobooth in case of failure of the raxpberry for exemple or use pictures on other support without open the photobooth  (sorry for my english )

anxuae, 2019-09-13:
> 
> ### Suggestion: add Webcam support
> 
> I would like to be able to use PiBooth also with a standard **Webcam** like Logitech C922 PRO.

@Wikinaut some webcam are now supported

JakubKontra, 2019-09-27:
What about CHROMAKEY support?

SBillion, 2019-10-18:
Hi,
First, big thanks for this project. I did a lot of party with a simple cardboard box to contain it. Hope to have more time to play a little bit more with it and do multiple size of box depending of the kind of party or festival. Lot of funny application are possible.
Anyway, I can't get the time to collaborate on your project at this moment but I need a simple way to generate all the photos again, using the originals present in raw folder and subfolders after changing the config file. This way, it would be easy for every pibooth owner to enhance their previous picture with a better title, more effects, etc...

werdeil, 2019-10-18:
> Hi,
> First, big thanks for this project. I did a lot of party with a simple cardboard box to contain it. Hope to have more time to play a little bit more with it and do multiple size of box depending of the kind of party or festival. Lot of funny application are possible.

If you can please send us photos of your setup, we will add them to the [wiki](https://github.com/werdeil/pibooth/wiki) section

> Anyway, I can't get the time to collaborate on your project at this moment but I need a simple way to generate all the photos again, using the originals present in raw folder and subfolders after [...]

I like the idea, I'll try to do an external script to rerun the photo generation, I've opened a dedicated issue #89 to follow this.

MickaelB, 2019-11-02:
Hi,

some features that can be usefull:
- hybrid mode support webcam (opencv)
- videobooth (webcam or raspi + micro) with ffmpeg or other
- webserver #52 

I'm a newbie in pyhon but i can help you in debug or others things

thanks

DEhrhardt, 2019-11-03:
I‘d like a full screen preview of the processed image for some seconds. Possibly instead of the thank you screen or additional.
Another possibility would be, that you can enlarge the last picture from the start screen by clicking or tapping.

I‘m already working on a gallery Webserver solution. I made it working with PicApport, but I didn’t really like that solution. I‘ll try Lychee next.

valerius-bits, 2019-11-10:
> Hi,
> 
> some features that can be usefull:
> 
> * hybrid mode support webcam (opencv)
> * videobooth (webcam or raspi + micro) with ffmpeg or other
> * webserver #52
> 
> I'm a newbie in pyhon but i can help you in debug or others things
> 
> thanks

Hi,
Second to those features 
+ preview image to be on screen all the time that waiting for selections
maybe a small window on right top

> * hybrid mode support webcam (opencv)
you can change in /controls/camera/___init___.py and hybrid.py  from "rpi_camera" to "cv_camera"

werdeil, 2019-11-12:
@MickaelB @valerius-bits when you are talking about "hybrid mode support webcam" do you want: 

- the picamera for preview and opencv camera for the capture?

or

- the opencv camera for the preview and a gphoto camera for the capture?

MickaelB, 2019-11-12:
Hi,
for me it's the opencv camera for the preview and a gphoto camera for the capture.

But someone could also need the the picamera for preview and opencv camera for the capture?

can we add 2 options in config ?
[Preview Driver] Rpicamera (default),opencv,gphoto2
[Capture Driver] Rpicamera (default),opencv,gphoto2

it's just a suggestion

thanks

Stephenrahill, 2019-12-21:
Hi,
Is their be a way of selecting the output image size? If not it would give great flexiablity to use odd paper sizes and specific websites without having to resize before uploading.

Thanks

werdeil, 2019-12-22:
Hi @Stephenrahill, it is quite difficult to have something working for all kind of output size, however I think it is a good idea to allow more than the 4x6 format, feel free to open an issue to ask for a specific format (square?), I'll see what I can do.

Exitras, 2020-03-16:
Hi, thank you very much for all your work! I love your project :)
I just finished my Photobooth and would love to add an external buzzer/trigger. Would it be possible to add a GPIO pin with a buzzer function i.e. that one picture is taken immediately (without countdown or selection of 1 or 4 pictures)?

anxuae, 2020-03-21:
Hi @Exitras ,

Thanks to the plugin system introduced by #115 , you will be able to drive any GPIO without modifying the core code of `pibooth`. This new functionality will be available soon in 2.X version.

anxuae

thibault-78, 2020-05-21:
> Hi,
> for me it's the opencv camera for the preview and a gphoto camera for the capture.
>
> But someone could also need the the picamera for preview and opencv camera for the capture?
>
> can we add 2 [...]

Hello All,
Thx for the great job you are doing
I'll aslo be intesreted to have the preview with a webcam and capture with gphoto2
I have tryed to modify the hybrid.py and init.py to make it works without any success
If anyone can help I will be happy to test

TheSuperpapi, 2020-09-15:
Hello there.
As someone already know, I get married this week end.
It was for me the first real time using the pibooth.

I want to thanks you guys for your great job, the pibooth was working quite very well.
I only had 3 pibooth shutdown that I need to investigate... it was just after picture shoot and before the last screen.

But It was ok using the reset switch installed on my box :)
[image]

450 pictures taken in 6hours
286 printed pictures
A lot of fun for all with some .... awasome and ... crazy pictures.

Thanks again again again

anxuae, 2020-09-15:
Hi @TheSuperpapi,
Your photo box is amazing! I like very much the casing!

If you have some feedbacks about the 3 shutdown, please send us (for instance, did you enabled the `animate` picture option?)

TheSuperpapi, 2020-09-15:
It's difficult to know.

Before my wedding time, I let the program started during 1.5 days without issue (just start without to use it).
Wanted to see if crash or hi RPI temperature.

Difficult for me to know exactly when it crash because people came to me after it happen...
They said me it happen after the pictures were taken.

What I need to investigate:
- SLR camera autofocus? (can create error if the camera is not able to stabilize with auto focus is on, but not sure because, in that case I have the oops screen) but don't think it's due to that.

- Maybe the memory get full and need to be empty? don't know if it can happen on a RPI like on windows.

For me, the frequency was quite the same, after more or less 1h30 hour each time.

So what i'll try:
- Taking picture for 1h30/ 3h00 time and try to catch exactly when the crash happen.
- Try to do that with an auto clear ram if I find how to.
- Give you a first feedback.

thanks for your comment about the casing,
My job is to build 3D models for automotive industries (industrial designer) and my neighbour job is to build 
 artistic prototype for city projects...

So, it was fun to build a high-finish case :)

Before and during the weeding time I had multiple proposal:
- The photographer ask me for a partnership
- The restaurant manager asked me if I could sell it
- Two couples asked me if I could rent it

Totally crazy...

For the investigation, if you have ideas, please don't hesitate :)
If you want some idea if improvement in term of new modules, I can back to you too, I catch a lot of comments.

See you soon,
Papi

Exitras, 2020-09-19:
Hi, I was having exactly the same experience as you. I used my PhotoBooth on two weddings so far using Pibooth 2.0.0 and experienced random shutdowns.
Maybe we can share our exact setups (in a structured way) to find similarities.

Raspberry Pi: RPi 4 with 4Gb ram.
Cooling fan on Raspberry Pi: yes
Original RPi USB-C PSU: yes 
Software Version: 2.0.0 (same issues with 1.x.x)
PiBooth LED feature used: No
Pictures additionally stored on USB: Yes
Touchscreen: No
Camera: Canon 2000D
Second camera: No 
Autofocus on camera: Off
Camera mode: Manual
External flash: No
Ferrite bead on camera connection cable: Yes
Printer: DNP DS620
Random crash: approximately after around 150 pictures taken but rather unpredictable.
Crash description: after Liveview (before taking the actual picture) PiBooth quits or goes to the „Uups screen“. 
Does the Pi also crash: No
Solution: restarting the System (power off - power on)

Lets use this structure and please add all things that might be relevant and I will add them in my post as well.

So far, I have added a cable with ferrite bead which seems decrease the crashes.

Greetings from Germany

TheSuperpapi, 2020-09-19:
Hello exitras,

I ha e stop using my current pibooth version and I have switched to the current dev version.

Stock with only pibooth, I have tried to reproduce the crash wothout succes.

From now i need to test it with my ds620 to see if it's due to the printing provess. 

So I'll come back on this thread later to inform you if the crash still occur. 

Anxuae informed me that they improved the logs system on the current dev version. 

Maybe I can advise you to test the current dev version in same time of me. 

Papi

Exitras, 2020-09-19:
Hey @TheSuperpapi this is good news :) Hope we will have a stable version soon 👍

TheSuperpapi, 2020-09-19:
Yes maybe but I just tried today,

I'll will add xscreensaver and setup cups later within the day or tomorrow and make new automatic test during 4h00 with rpinting sometime. 

I need to be sure that it'll not crash. 

So, if you find a process to produce the crash, d'o't hesitate. 

On my side to reproduce it on my previous version, I just had to force pibooth to take picture in like <forced mode> via vnc / windows and mouse bot auto click. 

So give me tomorrow and maybe next week to test it.

TheSuperpapi, 2020-09-19:
> Yes maybe but I just tried today,
>
> I'll will add xscreensaver and setup cups later within the day or tomorrow and make new automatic test during 4h00 with rpinting sometime.
>
> I need to be sure that [...]

anxuae, 2020-09-19:
Hi @Exitras and @TheSuperpapi, thanks for your feedback!
It could be really great to find the reason of this shutdown.

I have created the issue #182 to follow this particular topic.

twr412, 2020-09-24:
Has anyone changed the layout to a 2"x6" strip?

anxuae, 2020-09-26:
Hi @twr412 , you can use the [pibooth-picture-template](https://pypi.org/project/pibooth-picture-template/) plugin to customize the layout and the size

DJ-Dingo, 2020-10-13:
Adding a "User menu" / "Super user menu"
Sometimes i am not there to change things in the setup, so to prevent someone making a mistake it would be nice to have some kind of control. (maybe the chance to chosse what the user can change)

A User menu should only be:
--------------------
From Windows
    Text RGB color
    Flash on capture
    Preview delay

From Picture
    Orientation
    Number of captures
    Crop Captures
    Borders With
    Title
    Sub-title

The rest should be in some kind of "Super User Menu"
And maybe even a "SuperSuperUser Menu" for Debug Mode :-)

mmk8283, 2020-10-24:
Is there a way to display the final picture full screen? I know its been brought up, but I didn't find a solution. I would be happy just removing the "Take a picture" and arrow, and displaying the collage full screen.

mattdsteele, 2021-07-07:
Printing question: is there a way to automatically print a photo, and not display the prompt?

I'd prefer not to wire up an additional button or UI for folks, and just have a single print be automatic; similar to the way "traditional" photo booths work.

jasharrar, 2021-08-01:
> Hi Jasharrar,
> 
> Thanks a lot for your feedback. It's look really great!
> 
> anxuae

[image]

> Hi Jasharrar,
> 
> Thanks a lot for your feedback. It's look really great!
> 
> anxuae

[image]

New build for our wedding on Saturday
Is there a simple way for images to be copied to a usb stick as they are taken?

DJ-Dingo, 2021-08-01:
WOW it´s tall :-) 
@jasharrar You can add and extra quoted paths to an USB stick in the config.cfg 

[GENERAL]
# Path to save pictures (list of quoted paths accepted)
directory = ´~/Pictures/pibooth´, ´/paths to and USB stick/´

mattdsteele, 2021-08-16:
I built out and used my booth this weekend for a wedding reception. It worked great and was a hit with all the guests! 🎉 

Here's a blog post where I have photos and more details: https://steele.blue/photo-booth/

anxuae, 2021-08-17:
@mattdsteele very nice photobooth! Thank you for sharing!

dteplov182, 2021-08-28:
My version for Burning Man Russia. 

Inspired by https://www.thingiverse.com/thing:2121640, I have built the booth using 3 IKEA Lack tables. The walls are plexiglass.

[image]
[image]

anxuae, 2021-08-28:
Hi @dteplov182, thanks for sharing, and enjoy the pibooth!

Magicdaniel, 2021-09-01:
For me, this feature is missing: When i choose 4 photos, i would like to take 1 photo and have to press the button again for the next photo. Not all photos in a row. So the people in front of the box can reposition and dont have to hurry.

orikki, 2021-09-01:
I'm new so please forgive me. I really like the project and would like to donate some satoshis.
2nd is there a way to use a printer like the Mitsubishi CP-M15E or another roll-based printer?

anxuae, 2021-09-09:
Hi @Magicdaniel , thanks for your feedback.

> For me, this feature is missing: When i choose 4 photos, i would like to take 1 photo and have to press the button again for the next photo. Not all photos in a row. So the people in front of the box can reposition and dont have to hurry.
This will increase the time of a sequence, in my opinion, I prefer to redo a sequence sometime if people are not well positioned than request pressing a button at each capture. However you have 2 solution to improve user experience:
- increase the `preview_delay`
- create a plugin which wait for a press on a button

> And maybe the feature to show the preview all the time. So the people can position before they press the button for the picture.
It will be definitely out of the scope of this project, because the state concept becomes useless

anxuae, 2021-09-09:
Hi @orikki 

Thank you for donation proposal, until now we don't have opened a donate link.
Any printer working on linux should work with pibooth.

twisterrm, 2021-10-31:
Halloween kids party, 25 photo printed and approx other 50 not printed :)
I used some plugins, like google photo, sound effect (modified to suite my needs because the loop will not stop between the sequences), and picture template, the only thing i would suggest is to have a wiki or a doc for all the function, i had to use only dslr setup because if i use picamera and dslr the reflex camera was not used, only the picamera, and i haven't understand well the hybrid stuff.
BTW we have had a really fun time! Thanks again :)
[image]
[image]

anxuae, 2021-11-06:
Hi @twisterrm , thanks for the picture and your feedback. The documentation https://pibooth.readthedocs.io/en/latest/ is maybe to improve. Do not hesitate to open a issue with all the things that you didn't find in the doc

dominiquegrota, 2022-01-03:
Hi, 
Thank's to all your work, I could offer a Magic Mirror for my daughter's birthday

[image]
The magic mirror
[image]

[image]
Picture with flash leds on the mirror

[image]
Electric power and shutdown button, sound level button

[image]
The printer is vertically inside the mirror box. Paper and "ink" can be refilled without opening the box, a hole to turn on the printer.
I added the QRcode plugin and added a local wifi router inside, so everyone can connect to the wifi and get the picture (adding a apache server and a php script to download the picture)

The only feature I would suggest is to have an other led that bliks when the printing is validated, because I use a Canon selphy CP1300 and it needs 45 s to print the picture. When we used the magic miror at a party, some people did not wait for the picture, because they did not notice le printing icon on the screen (bottom left).

Thank's a lot !

aymonj, 2022-02-18:
Something interesting could be to have in configuration the ability to view the live cam.
I have the raspberry pi camera hq, and i always need to configure the camera with the commandline to adjust the aperture and focus. With this functionality, I don't need anymore the keyboard to run and configure the app.
Thanks a lot for your great work!

[47 more replies]