pibooth/pibooth #44: Touch screen not working only on home / full screen

m91192, 2018-09-19:
Hi project works like a charm thanks. I just cannot get how to set Start screen 'take photo' to be touched. In any size, mouse / touch / gpio does the job, but not in fullscreen.
Yet once this screen is passed, in full screen, all other screens work fine with touch...
help :)

jasharrar, 2018-09-19:
Hi I used these commit changes:
https://github.com/werdeil/pibooth/commit/b462ba9a41d58f03fa55f26443bb6a820e3e1492

anxuae, 2018-09-19:
Hello,

I fixed it in the commit that Jasharrar gives in it's comment. But it is not yet integrated in an official release.

@werdeil, maybe a new release can be done when the #43 is fixed?

anxuae

werdeil, 2018-09-23:
#43 is now fixed, feel free to do a new release ;-)

anxuae, 2018-09-25:
Fixed in ``pibooth`` 1.0.3

closed 2018-09-25 by anxuae