pibooth/pibooth #45: Add image effect selection in the config

anxuae, 2018-09-23:
Image effect could be defined in the config:

- picamera: use the attribute defined here: https://picamera.readthedocs.io/en/release-1.10/api_camera.html#picamera.camera.PiCamera.image_effect
- gphoto2: use the PIL library as defined here https://pillow.readthedocs.io/en/5.2.x/reference/ImageFilter.html (not all picamera filters available)
- possible to define a different effect for each capture (or one for all)

anxuae, 2018-09-24:
I made a first commit on the topic, see doc here: 
https://github.com/werdeil/pibooth#a-word-about-capture-effects .

However I can not test it for the moment (my raspberry is busy), you can test it by installing a developing version (see 
https://github.com/werdeil/pibooth#install-developing-version ). Maybe I could test next week.

anxuae

deha89, 2018-09-25:
After installing the developing-version I was able to add this line to pibooth --config:
effect = ("film", "cartoon","washedout", "film") And it works :-) During the preview you won't see the effect, but after merging, at the results-state you`ll see the effects.
Remark for others: When you want to take 4 Pictures, you have to define 4 effects.
@anxuae well done. Great Job! I will pay the next drink :-)

anxuae, 2018-09-25:
Fixed in ``pibooth`` 1.0.3

Please open new issues in case of bug/improvement

closed 2018-09-25 by anxuae