pibooth/pibooth #46: Disable Live-view for Sony DSLR

theonlyuser, 2018-09-26:
Hey,

yesterday I tested pibooth. 
I have a Sony DSLR and unfortunately the firmware does not support live-view. So 
I wanted to ask for the possibility, to use pibooth, without the preview function. It is not directly an issue from pibooth.

I testet gphoto2, capturing is working in console. With the sonyfirmware, unfortunally it is only possible to  capture an download the original taken photo, because it does not support the function to store it on  memory card.

manny thanks to the developers!

werdeil, 2018-10-01:
I updated a bit the issue title, I'll try to work on the behavior of the pibooth with a Sony DSLR, it may lead to disabling the live-view but it is not the only solution.

anxuae, 2019-07-21:
Setting parameters which are not supported don't raise exception (https://github.com/werdeil/pibooth/commit/ba37ca26b2d38284eb8305596e842da3fbbf8b18)

anxuae, 2019-07-21:
Think that preview using gphoto2 is relly painfull, the camera shall be initialized 2 times: ones before starting liveview+omxplayer, ones before capturing a new picture. Each initialization takes at least 3 seconds (imposed by gphoto2). Thus I think, liveview should not be used in pibooth in case of only a DSLR camera is connected.

anxuae, 2019-07-22:
Just a piste using `gphoto2-cffi` https://github.com/jbaiter/gphoto2-cffi/issues/2

anxuae, 2019-07-28:
Hi @teonlyuser,

I have worked on preview for DSLR camera, can you give me the model of your Sony camera?

Also please check which are its abilities on http://www.gphoto.org/proj/libgphoto2/support.php

Thanks
anxuae

theonlyuser, 2019-07-28:
Hey,

I m using a A6000. 

Kind regards

anxuae, 2019-07-29:
Ready for next release

commit d7a10eb9 (2019-07-28): Fix #46: rework DSLR preview to be optional
closed 2019-07-29 by anxuae