pibooth/pibooth #47: GUI to edit the configuration

anxuae, 2018-09-27:
A GUI can be made using `pygame` can be made to ease the configuration edition:

- maybe use the library https://github.com/ppizarror/pygame-menu
- develop it in a new python module (could be used in standalone mode to edit without starting `pibooth`)
- the call to this GUI can be done in the method `PiConfigParser.open_editor()`

werdeil, 2018-10-08:
I tried pygame-menu but it is quite a big menu with the current config file.
Maybe #50 will reduce it a bit but still.

anxuae, 2018-10-10:
Or we can consider that the GUI menu offer an easy way to configure `pibooth` only for the main keys and let the possibility to edit the config file for a full configuration.

Aurel-Charles, 2019-03-22:
I like the idea of having a quick config GUI menu : numbers of pics , fx, footers .....

but definitely keep a way to edit the config file for more keys as an expert mode config : for resolution, rotation of cam .... GPIO pin for buttons and LEDs

But its a good idea!!! :+1:

anxuae, 2019-04-15:
See first implementation https://github.com/werdeil/pibooth/commit/f1d9ccdb5fe5d785b45c27a9e6dfe86d71ab1643

closed 2019-06-10 by anxuae