pibooth/pibooth #48: Display the number of print waiting in the queue

werdeil, 2018-10-04:
I think it could be interesting to give the users a feedback on the number of print in the queue on cups side.

I'm thinking two options:
- Give the number on a new intermediate screen after pressing the print button
- Give the number on the main screen (small text at the bottom)

@anxuae what do you think?

anxuae, 2018-10-07:
A quick solution is to add indication in the print screen and the main screen

werdeil, 2018-10-08:
Yes but I'm really balanced between this and a new intermediate screen (and state) where we show the printing acknowledgement and queue

anxuae, 2018-10-10:
I think that we can use the existing screens (and state), the information can be really useful at the wait state. A new state will be short-lived and thus may be a little bite less interesting.

anxuae, 2018-10-27:
First implementation 
https://github.com/werdeil/pibooth/commit/e97098ef06bfb7c43a9e74cbaa7aad10653cf487

deha89, 2018-11-21:
The implementation[ e97098e](https://github.com/werdeil/pibooth/commit/e97098ef06bfb7c43a9e74cbaa7aad10653cf487) works well :-)

anxuae, 2018-12-06:
Thanks for your feedback, I close the issue.
We will manage improvements in new tickets, if required.

closed 2018-12-06 by anxuae