pibooth/pibooth #49: Hide arrows when a touchscreen is user

anxuae, 2018-10-06:
It could be interesting to hide the arrows (indicating the HW buttons) when a touchscreen is used as they are useless.

jasharrar, 2018-10-06:
I managed to do this by just editing the images in paint.  It would be good to have options so that you could choose this as an option on install.

anxuae, 2018-10-07:
Can't find an easy way to detect if an touchscreen is used.
We can add a new key in the config, but this is one more...

An other way is the set empty value to `picture_btn_pin` or `print_btn_pin`.

Don't know what is the best option

anxuae, 2018-10-07:
finally I added a new key in the config 
https://github.com/werdeil/pibooth/commit/17548529a1b6a6c6536749aaa33d6569971a469b

closed 2018-10-07 by anxuae