pibooth/pibooth #50: Optimize the number of keys in the config

anxuae, 2018-10-07:
There are lots of keys in the config. Some one can be removed to ease the `pibooth` configuration (e.g. `arrows`).

Alex74300, 2018-10-17:
Bonjour Vincent

Je souhaiterai modifier le footer des impressions pibooth en y mettant une image png à la place du footer_text. Cela est il possible en modifiant le fichier config ou un fichier .py?
Merci 
Bonne journée

anxuae, 2019-04-15:
We can close this ticket, the GUI is a good option to have a light weightconfig

werdeil, 2019-04-16:
No I don't think we should close it, we should be able to remove some of the keys and arrows is a good example I think.

anxuae, 2019-04-16:
Typically for arrows, I don't know how to detect the touchscreen (and don't display them)

anxuae, 2020-03-23:
Keys to be removed from config:

- clear_on_startup

werdeil, 2020-04-01:
> Keys to be removed from config:
> 
> * clear_on_startup

Removed in e758c45

anxuae, 2021-04-23:
@werdeil do you think that this issue shall be kept opened?

werdeil, 2021-05-25:
Closing it as we won't be able to fix it the way I wanted and I know you don't like having issue opened. I'll may find a workaround some day.

closed 2021-05-25 by werdeil