pibooth/pibooth #51: picture in the footer pibooth

Alex74300, 2018-10-17:
Bonjour Vincent

Je souhaiterai modifier le footer des impressions pibooth en y mettant une image png à la place du footer_text. Cela est il possible en modifiant le fichier config ou un fichier .py?
Merci
Bonne journée

werdeil, 2018-10-17:
Bonjour,

C'est déjà possible, pour cela il faut utiliser une image pour l'arrière plan au lieu des footer texts.

Il suffit donc de créer une image de background au bon format et ensuite de mettre le chemin vers l'image dans la clé bg_color de la configuration au lieu d'un triplé RGB.

Est ce que cela répond à votre besoin?

Alex74300, 2018-10-17:
oui merci! mais du coup ou trouve on cette clé bg_color ? dois je modifier le fichier py ou le cfg? quel est la forme de l'expression et comment identifier le chemin path? avez vous un exemple de formulation? 
merci !
alx

werdeil, 2018-10-17:
Il vous faut modifier le fichier config qui est présent suivant le chemin suivant ~/.config/pibooth/pibooth.cfg, ou simplement en lançant pibooth --config.

Dans la clé bg_color, mettre le chemin vers votre fichier d'image par exemple:
bg_color = ~/chemin/vers/votre/image/background

anxuae, 2018-10-19:
Can we close this issue?

Alex74300, 2018-10-19:
Merci ! oui grâce à vos indications, j'ai réussi à faire ce que je souhaitais ! 
bonne soirée

closed 2018-10-19 by Alex74300