pibooth/pibooth #52: Publish pictures Pibooth

Alex74300, 2018-11-20:
Hello

I'm looking for a solution to use photos taken with Pibooth and publish them on my Facebook or Instagram profile or send them to a cloud (online folder) or through a QR code integration. Is there a way to integrate command lines to your program to have this feature via a dedicated GPIO button? Thank you !

Aurel-Charles, 2019-03-22:
I've been looking for something similar, but I really have no idea of how to do it .
My idea would be to have one setting to either send or store  pictures in a cloud folder (via a account I guess) , so people just have to access this through a link.

Here is some stuff I found on internet that might be a lead:

https://github.com/wyolum/TouchSelfie
https://github.com/laurentalacoque/TouchSelfie-extended/wiki/CreateGoogleProject

maybe it can help!

Aurel

anxuae, 2019-10-29:
Hi @DEhrhardt ,

Thanks for your post, we have imagined the same solution but never succeed.
Have a look to https://github.com/werdeil/pibooth/blob/master/docs/lychee.rst 
Maybe we can improve the doc with your feedback.

anxuae

DaddioD, 2020-01-29:
I have been experimenting with a few things to accomplish this.  The following works so far.

1. Take picture(s) with pibooth
2. have python script in crontab that watches folder and sends images with a label added to gmail when new ones are added.
3. Apps script in gmail adds attachments with label to folder in google drive.

So far this is just in testing but has worked for a kids party that generated 46 different final images.

dvid9976, 2020-02-04:
hi all

I got some basic "sharing" by using andreafabrizi's Dropbox-Uploader script. 

Basically connected the script to a shared app in dropbox and added a crontab job to upload pibooth images folder every 2 minutes.

You can share the dropbox folder to whoever you want and from there.. it is "their thing" :-)

I'm not a developer and this is not too much, but it is enough to me. Hope it helps

anxuae, 2020-03-19:
Example on how to upload pictures on a FTP server will be provided with the new plugin system.

anxuae, 2020-03-24:
New plugin system will let you the possibility to add custom code to upload pictures.
Will be available in next `pibooth` major release (date to be planned).
See plugins example at https://github.com/werdeil/pibooth/blob/master/docs/plugin.rst .

closed 2020-03-24 by anxuae