pibooth/pibooth #53: Error -53 Could not claim the USB device

giovannimartusciello, 2018-12-27:
Hi, I installed everything on the raspberry, when I try to take a photo I get the error -53 or failed Resource busy..
seems really strange because if I go out og pibooth and take a pick only using gphoto it works.
Any suggestment?

anxuae, 2019-01-06:
Hi,

It's difficult to reproduce this issue without more info, Can you give me:

- Model of your DSLR camera
- Full python traceback (set in the `pibooth` config `failsafe = True` before running the programme)
- The gphoto command that you used?

loloNeuch, 2019-02-06:
Hi,

Is there and feedback about this issue?
I have the same error. I can use gphoto directly but tgrough pibooth I have the error mentionnes above.
I use a Canon PowerShot G10 as camera.
Thanks in advance!

anxuae, 2019-02-12:
Please can you paste the traceback generated when `failsafe = True` in the config?
Can not investigate without more information (and I have not a Canon PowerShot G10).

layer91, 2019-02-15:
This is very much still an issue! Was hoping to use the photobooth tomorrow, but right now i can't proceed. I HAVE to manually stop the program when it fails, as it otherwise will MELT the sdcard, which has already happened... 
The camera opens the shutter, but never proceeds to close it afterwards, which is why i have to manually disconnect the camera to avoid burning the card. 

I have tried setting permissions for gvfs-gphoto-volume-monitor to 0 in /usr/lib/gvfs
I also tried gvfs-related files and mount points according to https://github.com/reuterbal/photobooth/blob/master/INSTALL.md
sudo rm /usr/share/dbus-1/services/org.gtk.vfs.GPhoto2VolumeMonitor.service
sudo rm /usr/share/gvfs/mounts/gphoto2.mount
sudo rm /usr/share/gvfs/remote-volume-monitors/gphoto2.mount
sudo rm /usr/lib/gvfs/gvfs-gphoto2-volume-monitor
sudo rm /usr/lib/gvfs/gvfsd-gphoto2 
This has no or the same effect as disabling read/write as above. 

I have tried using GPhoto2 as provided by Apt-get, and both the stable and dev version from the script provided in the installation instructions. 
First attached stacktrace is with config option preview_stop_on_capture = False
The second trace is with preview_stop_on_capture = False.
in both cases i turn the camera off after it fails to avoid burning the sd. 

pi@raspberrypi:~ $ pibooth
[ INFO    ] pibooth           : Starting the photo booth application...
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 pictures selected)
[ INFO    ] pibooth           : Start new pictures sequence
Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.
[ INFO    ] pibooth           : Take picture and save it in /home/pi/Pictures/pibooth/raw/2019-02-15-13-07-27/pibooth000.jpg
[ WARNING ] gphoto2           : (b'gp_libusb1_open [libusb1.c:421]') b"'libusb_claim_interface (port->pl->dh, port->settings.usb.interface)' failed: Resource busy (-6)"
[ WARNING ] gphoto2           : (b'gp_port_set_error [gphoto2-port.c:1190]') b'Could not claim interface 0 (Success). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.'
[ ERROR   ] pibooth           : [-53] Could not claim the USB device
Ctrl-C pressed ... Exiting.
Movie capture finished (41 frames)
^C[ WARNING ] gphoto2           : (b'gp_port_set_error [gphoto2-port.c:1190]') b'Could not release interface 0 (No such device).'
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.5/dist-packages/pibooth/booth.py", line 523, in main
    app.main_loop()
  File "/usr/local/lib/python3.5/dist-packages/pibooth/booth.py", line 475, in main_loop
    clock.tick(40)  # Ensure the program will never run at more than x frames per second
KeyboardInterrupt

Output from lsusb -v
Bus 001 Device 007: ID 04b0:0421 Nikon Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x04b0 Nikon Corp.
  idProduct          0x0421 
  bcdDevice            1.00
  iManufacturer           1 NIKON
  iProduct                2 NIKON DSC D90
  iSerial                 3 000006585872
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           39
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower                2mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
[... 142 lines cut ...]
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c0 receiving resp failed: PTP Device Busy (0x2019)'
[... cut ...]
[ INFO    ] pibooth           : Start new pictures sequence
Capturing preview frames as movie to 'stdout'. Press Ctrl-C to abort.

*** Error ***              
An error occurred in the io-library ('Could not claim the USB device'): Could not claim interface 0 (Device or resource busy). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module (such as sdc2xx, stv680, spca50x) is using the device and you have read/write access to the device.
ERROR: Movie capture error... Exiting.
Movie capture finished (0 frames)

layer91, 2019-02-15:
I found a script to test the camera i have on Gphotos github. Perhaps it contains something useful?
pi@raspberrypi:~ $ ./test-nikon-d90.sh 
Attach Nikon D90 and press return: 
echo "***  basics"
++ echo '***  basics'
***  basics
gphoto2 -L
++ gphoto2 -L
There is no file in folder '/'.                                                
There is no file in folder '/store_00010001'.
There is no file in folder '/store_00010001/DCIM'.
There are 23 files in folder '/store_00010001/DCIM/100NCD90'.
#1     DSC_0001.JPG               rd  3710 KB 4288x2848 image/jpeg
#2     DSC_0002.JPG               rd  3472 KB 4288x2848 image/jpeg
#3     DSC_0003.JPG               rd  3200 KB 4288x2848 image/jpeg
#4     DSC_0004.JPG               rd  3614 KB 4288x2848 image/jpeg
#5     DSC_0005.JPG               rd  4151 KB 4288x2848 image/jpeg
#6     DSC_0006.JPG               rd  3671 KB 4288x2848 image/jpeg
#7     DSC_0007.JPG               rd  3407 KB 4288x2848 image/jpeg
#8     DSC_0008.JPG               rd  3436 KB 4288x2848 image/jpeg
#9     DSC_0009.JPG               rd  3421 KB 4288x2848 image/jpeg
#10    DSC_0010.JPG               rd  4573 KB 4288x2848 image/jpeg
#11    DSC_0011.JPG               rd  4613 KB 4288x2848 image/jpeg
#12    DSC_0014.NEF               rd  9680 KB application/x-unknown
#13    DSC_0015.JPG               rd  3264 KB 4288x2848 image/jpeg
#14    DSC_0018.NEF               rd  8879 KB application/x-unknown
#15    DSC_0019.JPG               rd  3362 KB 4288x2848 image/jpeg
#16    DSC_0020.JPG               rd  3343 KB 4288x2848 image/jpeg
#17    DSC_0021.JPG               rd  3349 KB 4288x2848 image/jpeg
#18    DSC_0022.JPG               rd  3336 KB 4288x2848 image/jpeg
#19    DSC_0023.JPG               rd  3301 KB 4288x2848 image/jpeg
#20    DSC_0024.JPG               rd  3827 KB 4288x2848 image/jpeg
#21    DSC_0025.JPG               rd  3833 KB 4288x2848 image/jpeg
#22    DSC_0026.JPG               rd  3655 KB 4288x2848 image/jpeg
#23    DSC_0027.JPG               rd  3520 KB 4288x2848 image/jpeg
gphoto2 -l
++ gphoto2 -l
There is 1 folder in folder '/'.                                               
 - store_00010001
There is 1 folder in folder '/store_00010001'.
 - DCIM
There is 1 folder in folder '/store_00010001/DCIM'.
 - 100NCD90
There are 0 folders in folder '/store_00010001/DCIM/100NCD90'.
gphoto2 --summary
++ gphoto2 --summary
Camera summary:                                                                
Manufacturer: Nikon Corporation
Model: D90
  Version: V1.00
  Serial Number: 6585872
Vendor Extension ID: 0xa (1.0)
Vendor Extension Description: microsoft.com: 1.0

Capture Formats: JPEG Undefined Type
Display Formats: JPEG, Undefined Type, Association/Directory, MS AVI, DPOF, Script

Device Capabilities:
	File Download, File Deletion, File Upload
	Generic Image Capture, No Open Capture, Nikon Capture 1, Nikon Capture 2

Storage Devices Summary:
store_00010001:
	StorageDescription: None
	VolumeLabel: pictures   
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read Only with Object deletion
	Maximum Capability: 15914401792 (15177 MB)
	Free Space (Bytes): 15817433088 (15084 MB)
[... 434 lines cut ...]
New file is in location /store_00010001/DCIM/100NCD90/DSC_0038.NEF on the camera
Saving file as DSC_0038.NEF                                                    
Deleting file /store_00010001/DCIM/100NCD90/DSC_0038.NEF on the camera
Capturing frame #2/3...
New file is in location /store_00010001/DCIM/100NCD90/DSC_0039.JPG on the camera
Saving file as DSC_0039.JPG                                                    
Deleting file /store_00010001/DCIM/100NCD90/DSC_0039.JPG on the camera
Waiting for next capture slot 2 seconds...
New file is in location /store_00010001/DCIM/100NCD90/DSC_0039.NEF on the camera
Saving file as DSC_0039.NEF                                                    
Deleting file /store_00010001/DCIM/100NCD90/DSC_0039.NEF on the camera
Capturing frame #3/3...
New file is in location /store_00010001/DCIM/100NCD90/DSC_0040.JPG on the camera
Saving file as DSC_0040.JPG                                                    
Deleting file /store_00010001/DCIM/100NCD90/DSC_0040.JPG on the camera
New file is in location /store_00010001/DCIM/100NCD90/DSC_0040.NEF on the camera
Saving file as DSC_0040.NEF                                                    
Deleting file /store_00010001/DCIM/100NCD90/DSC_0040.NEF on the camera
gphoto2 --set-config imagequality=2
++ gphoto2 --set-config imagequality=2
                                                                               
echo -n "interval capture done ... press return to continue"
++ echo -n 'interval capture done ... press return to continue'
interval capture done ... press return to continueread dummy
++ read dummy

layer91, 2019-02-15:
I noticed [ WARNING ] pibooth           : Invalid value 'Carte mémoire' for option capturetarget (possible choices: ['Internal RAM', 'Memory card']), still trying to set it
in the logs. Sounds like french??

werdeil, 2019-02-15:
Hi @layer91 

Can you try to test on the version 1.0.1? https://github.com/werdeil/pibooth/releases/tag/1.0.1

Since the version 1.0.2 we've try to deliver a better preview which works well on my Canon DSLR but this may be the reason it doesn't work with your DSLR.

And about the Invalid value 'Carte mémoire' it is indeed french for 'Memory Card', as I'm french and my DSLR is configured in french I needed it to make it work. You can try to change the line 244 of pibooth/controls/camera.py to 
`gp_set_config_value(config, 'settings', 'capturetarget', 'Memory Card')`

layer91, 2019-02-15:
I get the following error straight away. 
pi@raspberrypi:~/pibooth-1.0.1 $ pibooth
[ INFO    ] pibooth           : Starting the photo booth application...
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'ISO Auto Hi Limit' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'High ISO Noise Reduction' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Flash Exposure Compensation' expected: 0x0002 got: 0x0001"
[ WARNING ] pibooth           : Invalid value '200' for option iso (possible choices: ['100', '125', '160', '200', '250', '320', '400', '500', '640', '800', '1000', '1250', '1600', '2000', '2500', '3200', '4000', '5000', '6400'])
[ WARNING ] pibooth           : Invalid value 'Memory Card' for option capturetarget (possible choices: ['Internal RAM', 'Memory card'])

layer91, 2019-02-15:
pi@raspberrypi:~/pibooth-1.0.1 $ sudo gphoto2 -v
gphoto2 2.5.20.1

Copyright (c) 2000-2019 Lutz Mueller and others

gphoto2 comes with NO WARRANTY, to the extent permitted by law. You may
redistribute copies of gphoto2 under the terms of the GNU General Public
License. For more information about these matters, see the files named COPYING.

This version of gphoto2 is using the following software versions and options:
gphoto2         2.5.20.1       gcc, popt(m), exif, no cdk, no aa, jpeg, no readline
libgphoto2      2.5.12         all camlibs, gcc, ltdl, EXIF
libgphoto2_port 0.12.0         gcc, ltdl, USB, serial without locking

layer91, 2019-02-15:
Hey! Update! Downgraded to 1.0.1 and it works, kind of... First round goes well, after that it errors. Log attached.
pi@raspberrypi:~ $ pibooth
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Starting the photo booth application...
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'ISO Auto Hi Limit' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'High ISO Noise Reduction' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Flash Exposure Compensation' expected: 0x0002 got: 0x0001"
[ WARNING ] pibooth           : Invalid value '100' for option iso (possible choices: ['100', '125', '160', '200', '250', '320', '400', '500', '640', '800', '1000', '1250', '1600', '2000', '2500', '3200', '4000', '5000', '6400'])
[ WARNING ] pibooth           : Invalid value 'Carte mémoire' for option capturetarget (possible choices: ['Internal RAM', 'Memory card'])
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 pictures selected)
[ INFO    ] pibooth           : Start new pictures sequence
[ INFO    ] pibooth           : Capturing new preview image
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Take picture and save it in /home/pi/Pictures/pibooth/2019-02-15-18-21-11/ptb000.jpg
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'camera_nikon_capture [library.c:3093]') b'Parentobject of newobject 0x2f000001 is 0x0 now?'
[ INFO    ] pibooth           : Creating merged picture
[ INFO    ] pibooth           : Save the merged picture in /home/pi/Pictures/pibooth/2019-02-15-18-21-11/2019-02-15-18-21-26_ptb.jpg
[ INFO    ] pibooth           : Display the merged picture
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 pictures selected)
[ INFO    ] pibooth           : Start new pictures sequence
[ INFO    ] pibooth           : Capturing new preview image
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x9201 receiving resp failed: Invalid Status (0xa004)'
[ WARNING ] gphoto2           : (b'camera_capture_preview [library.c:2683]') b"'ret' failed: Nikon enable liveview failed (0xa004: Invalid Status)"
[ WARNING ] gphoto2           : (b'gp_context_error') b'Nikon enable liveview failed (0xa004: Invalid Status)'
[ WARNING ] gphoto2           : (b'gp_camera_capture_preview [gphoto2-camera.c:1404]') b"'camera->functions->capture_preview ( camera, file, context)' failed: -1"
[ ERROR   ] pibooth           : [-1] Unspecified error

layer91, 2019-02-15:
Another update!
The issue with version 1.0.1 was because the camera was set to output BOTH raw and JPEG, which confused the program, causing an error before the next photo. Maybe mention to check those settings somewhere? haha! 
I tried the latest version, but that still triggers the same original message,

werdeil, 2019-02-15:
As latest version is not really bugproof for all DSLR (as gphoto2 doesn't have the same development for all DSLR) I will work tomorrow on a version without preview (only photo) when using only a DSLR. The best version with preview is the one with the PiCamera , either the PiCamera only or the hybrid mode with PiCamera and DSLR.

anxuae, 2019-03-09:
Fix pushed to translate DSLR config values in appropriate language

closed 2019-07-20 by anxuae