pibooth/pibooth #54: Crash on 4grid

layer91, 2019-02-15:
Hi! Ran into an issue with the 4 photos 2x2 grid option.
1,2 and 3 execute without issue, option 4 causes Oopsie

Logs
pi@raspberrypi:~/pibooth-1.0.1 $ pibooth
[ INFO    ] pibooth           : Starting the photo booth application...
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'ISO Auto Hi Limit' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'High ISO Noise Reduction' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Flash Exposure Compensation' expected: 0x0002 got: 0x0001"
[ WARNING ] pibooth           : Invalid value '100' for option iso (possible choices: ['100', '125', '160', '200', '250', '320', '400', '500', '640', '800', '1000', '1250', '1600', '2000', '2500', '3200', '4000', '5000', '6400'])
[ WARNING ] pibooth           : Invalid value 'Carte mémoire' for option capturetarget (possible choices: ['Internal RAM', 'Memory card'])
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (4 pictures selected)
[ INFO    ] pibooth           : Start new pictures sequence
[ INFO    ] pibooth           : Capturing new preview image
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Take picture and save it in /home/pi/Pictures/pibooth/2019-02-15-21-36-59/ptb000.jpg
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'camera_nikon_capture [library.c:3093]') b'Parentobject of newobject 0x2b000008 is 0x0 now?'
[ INFO    ] pibooth           : Capturing new preview image
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Take picture and save it in /home/pi/Pictures/pibooth/2019-02-15-21-36-59/ptb001.jpg
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x9201 receiving resp failed: Invalid Status (0xa004)'
[ WARNING ] gphoto2           : (b'camera_nikon_capture [library.c:3093]') b'Parentobject of newobject 0x2b000009 is 0x0 now?'
[ INFO    ] pibooth           : Capturing new preview image
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Take picture and save it in /home/pi/Pictures/pibooth/2019-02-15-21-36-59/ptb002.jpg
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x9201 receiving resp failed: Invalid Status (0xa004)'
[ WARNING ] gphoto2           : (b'camera_nikon_capture [library.c:3093]') b'Parentobject of newobject 0x2b00000a is 0x0 now?'
[ INFO    ] pibooth           : Capturing new preview image
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Capturing new preview image
[ INFO    ] pibooth           : Take picture and save it in /home/pi/Pictures/pibooth/2019-02-15-21-36-59/ptb003.jpg
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x9201 receiving resp failed: Invalid Status (0xa004)'
[ WARNING ] gphoto2           : (b'camera_nikon_capture [library.c:3093]') b'Parentobject of newobject 0x2b00000b is 0x0 now?'
[ INFO    ] pibooth           : Creating merged picture
[ WARNING ] gphoto2           : (b'gp_context_error') b'You need to specify a folder starting with /store_xxxxxxxxx/'
[ WARNING ] gphoto2           : (b'gp_camera_file_get [gphoto2-camera.c:1693]') b"'gp_filesystem_get_file (camera->fs, folder, file, type, camera_file, context)' failed: -1"
[ ERROR   ] pibooth           : [-1] Unspecified error

werdeil, 2019-02-15:
I got the same issue when there were not enough space on the internal memory of the camera, hence the setting to use the memory card instead of the internal memory. Have you tried to change the line as explained in the other issue thread?

I hope this helps, and I'm sorry I could not give you a better answer.

layer91, 2019-02-16:
I've been a bit scared to use the sd, considering the meltdown of the last.
On this run of testing (literally done this for 14 hours today...) I went back to a clean version of 1.0.1. 
I'll try adjusting the error from Carte memoire to Memory card tomorrow. 
[ WARNING ] pibooth : Invalid value 'Carte mémoire' for option capturetarget (possible choices: ['Internal RAM', 'Memory card'])
Thanks for your help!

anxuae, 2019-02-16:
The configuration of the DSL camera seems to be language dependent, will be hard to fix that (doing a mapping for each config key and supported language?)

anxuae, 2019-03-09:
Fix pushed to translate DSLR config values in appropriate language

closed 2019-07-21 by anxuae