pibooth/pibooth #55: Set the border around images

layer91, 2019-02-15:
Is there any way to reduce the border/padding that ends up in the print? Both left, right and bottom, to basically fill the paper as best possible. 
Footer test is set to nothing. 
Is there anything that can be done about this, or have i simply missed something?
It seems no matter the config I cannot do anything about the padding 

See attached image. Orientation is set to auto in both cases. 
 
These are 2 square images. 2848, 2848. I think this would be better in landscape
[image]

This is two images with resolution 4288x2848 This would be better in portrait mode I think
[image]

werdeil, 2019-02-15:
Normally if you put no footer text it shall use the whole height of the photo to put the images. Can you confirm that you kept the "footer_text1" and "footer_text2" keys in the config but put empty value in it (like "" or None)?

I don't think it is a good idea to give the possibility to set the border size around images.

layer91, 2019-02-16:
A more dynamic way to set up to layout would be amazing, but probably out of scope. 
The variables are set as
footer_test1 = 
footer_test2 = 
Should it rather be a double quote "" or the word None?

anxuae, 2019-02-16:
Hi,

I'have pushed a fix in order to improve the `orientation=auto` mode, but this doesn't change the padding .

The `footer_test1` and `footer_test2` shall be empty (not None nor "").

anxuae

layer91, 2019-02-16:
I'm glad to hear! We're already live now so, it is what it is. Works great with picam as preview tho! Sent from my Huawei Mobile-------- Original Message --------Subject: Re: [werdeil/pibooth] Set the border around images (#55)From: Tony Anxuae To: werdeil/pibooth CC: layer91 ,Author Hi,
I'have pushed a fix in order to improve the orientation=auto mode, but this doesn't change the padding .
The footer_test1 and footer_test2 shall be empty (not None nor "").
anxuae

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/werdeil/pibooth/issues/55#issuecomment-464361451",
"url": "https://github.com/werdeil/pibooth/issues/55#issuecomment-464361451",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

anxuae, 2019-03-09:
Ok, it seems difficult to add borders handling in config

closed 2019-03-09 by anxuae