pibooth/pibooth #57: I am not sure what is the issue with my Pipooth

kboghdady, 2019-03-09:
Here is my logs 
[ INFO    ] pibooth           : Starting the photo booth application...
[ INFO    ] pibooth           : Connected to printer 'hp_LaserJet_4200_(0001E68EF278)_via_VNC_from_Kiro-MKP-2'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 pictures selected)
[ INFO    ] pibooth           : Start new pictures sequence
timer 3
timer 2
timer 1
[ INFO    ] pibooth           : Take picture and save it in /root/Pictures/pibooth/raw/2019-03-08-19-03-15/pibooth000.jpg
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Lens Name' expected: 0x0002 got: 0x0004"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002"
[ WARNING ] gphoto2           : (b'_get_config [config.c:7471]') b"Type of property 'Live View AF Mode' expected: 0x0001 got: 0x0002"
[ WARNING ] pibooth           : Invalid value 'Carte mémoire' for option capturetarget (possible choices: ['Internal RAM', 'Memory card']), still trying to set it
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[ WARNING ] gphoto2           : (b'camera_nikon_capture [library.c:3093]') b'Parentobject of newobject 0xb000007 is 0x0 now?'
[ WARNING ] gphoto2           : (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x100b receiving resp failed: PTP Invalid Object Handle (0x2009)'
[ WARNING ] gphoto2           : (b'camera_nikon_capture [library.c:3125]') b'deleteobject(b000007) failed: 2009'
[ WARNING ] gphoto2           : (b'camera_nikon_capture [library.c:3131]') b'deleteobjectinsdram(b000007) failed too: 2009'
[ INFO    ] pibooth           : Creating merged picture
[ WARNING ] gphoto2           : (b'gp_camera_file_get [gphoto2-camera.c:1693]') b"'gp_filesystem_get_file (camera->fs, folder, file, type, camera_file, context)' failed: -108"
[ ERROR   ] pibooth           : [-108] File not found

anxuae, 2019-03-09:
Hi,

It is probably the same issue than #53.
We have let some code in french to configure the DSLR camera, and you are probably using a camera with english language.

I can fix it by adapting the language used to configure the camera according to the one chosen in the configuration file.

I try to push fix it today.

anxuae

anxuae, 2019-03-09:
Fix pushed to translate DSLR config values in appropriate language

commit 6058b1cc (2019-03-09): Fix #57, #54, #53
closed 2019-03-21 by kboghdady