pibooth/pibooth #58: Problem with new update

Alex74300, 2019-03-15:
Hi

I updated yesterday the Pibooth app and I now have a problem with buttons 1 and 2 when printing. I think the new function "Choosing the layout with the buttons 1 and 2" create a conflict with the print function. Both buttons 1 and 2 send a photo to print, whereas before only button 2 allowed printing. Button 1 reset the program and returned to the layout screen.
How can you change the problem so that each button has a specific function?
Thank you for your help!
alex

Alex74300, 2019-03-15:
It's ok ! I unistall and update agin and it work !
thanks !

anxuae, 2019-03-15:
Hi Alex,

Can you publish your `pibooth` config?

BR
anxuae

anxuae, 2019-04-21:
Hi,
Some news about this issue ?
Thanks

anxuae, 2019-06-09:
Can not reproduce. Without logs, I will close this issue in some days.

Regards.

Anxuae

closed 2019-06-12 by anxuae