pibooth/pibooth #61: Preview stop on capture triggers AttributeError in hybrid mode

etiennedm, 2019-04-17:
When running in hybrid mode and configure with preview_stop_on_capture = True, I get the following error:

`Traceback (most recent call last):
  File "/home/pi/pibooth/pibooth/states.py", line 68, in process
    self.active_state.do_actions(events)
  File "/home/pi/pibooth/pibooth/booth.py", line 210, in do_actions
    self.app.camera.capture(capture_path, effect)
  File "/home/pi/pibooth/pibooth/controls/camera.py", line 427, in capture
    rect = self.get_rect()
  File "/home/pi/pibooth/pibooth/controls/camera.py", line 118, in get_rect
    rect = self._window.get_rect()
AttributeError: 'NoneType' object has no attribute 'get_rect'`

This is due to the fact that `stop_preview` is called in booth.py:193, in which we set `self._window = None`, causing the following call to `get_rect` in `capture` (in booth.py:210) to fail.

I don't know how you would like to fix this. Pass a `capture_stopped` argument to `capture`, or keep track of capture state in RpiCamera/GpCamera (probably cleaner).

werdeil, 2019-04-18:
My bad this is my new development of the "smile" screen in camera line 425

I would just put this on a "if self._window" what do you think @anxuae ?

anxuae, 2019-04-20:
Yes, self._window is None when preview is closed or deactivated. Thus no smile should be displayed.

Also I think that this part of code should be in RpiCamera.

commit 90bbb999 (2019-06-09): Fix #61: stop on capture in hybrid mode
closed 2019-06-09 by anxuae