pibooth/pibooth #62: Feature- print count for canon selphy

jasharrar, 2019-05-21:
Hi 
I'm using a canon selphy so need to change ink every 38 prints and paper every 19 prints. Is there some way to have a count so a message pops up to change the media?

Thanks

werdeil, 2019-05-21:
Hi, I understand the need but until now there is no way for the pibooth to know the status of the printer at startup, so we cannot know how many prints are left.

Until now the whole printing part is using cups in order to avoid handling each brand of printer. I don't see a way to change it easily.

@anxuae ?

jasharrar, 2019-05-22:
I thought that may be the case, bit could a separate counter be added in the config as a work around. This would need to be "reset" manually by exiting pibooth and loading the config.

anxuae, 2019-06-09:
Can have a look to CUPS attributes: https://www.cups.org/doc/spec-ipp.html
It seems possible to get the ink level if the printer shall the info with the CUPS server.

Otherwise a counter can be implemented.

anxuae, 2019-07-28:
Hi @jasharrar,

I will add a counter in the config file, it will be displayed each time a new picture is sent to the printer.

Do you think that if the counter is reached, all print actions shall be blocked until counter reset?

anxuae

jasharrar, 2019-07-28:
That would be really helpful, alternatively some kind of GUI message to replace the paper/ink. I think a lot of us are using cannon selphys. With 19 prints per pack and 38 prints per ink roll.

anxuae, 2019-07-29:
Hi @jasharrar,

Can you make a test for me?

Connect your printer to your raspberry pi, power-on rpi&printer.
Then in a python shell, execute these lines:
```
import cups

p=cups.Connection()
attributes = p.getPrinterAttributes('<name of your printer>')

for name in attributes:
    if 'marker' in name:
    print(name, attributes[name])

```

Then can you post the result?
I would like to see it is possible to get the remaining supply levels of the printer.

Thanks

closed 2019-08-24 by anxuae

anxuae, 2019-08-24:
New key 'max_pages' in config to define the printer counter.
The counter can be reset by opening/closing the settings menu.

mattdsteele, 2021-07-07:
@jasharrar I know it's been a while, but I'm curious how your experience with the Canon Selphy went?

I'm considering picking one up for my photo booth project, but want to make sure it prints OK, you didn't have to do anything weird to configure it on the Pi, etc.

Thanks! 🙌

sandrokeil, 2021-07-25:
@mattdsteele I can confirm that printing works with Canon Selphy CP1300 and this awesome app. I use a Raspberry PI 4 with 4GB RAM and Raspberry Pi OS Lite. After some configuration I got all the things working.

dominiquegrota, 2021-11-28:
Hi, I confirm that your app is excellent !!! and it works with cp1300 printer. 
The [PRINTER][max_pages] count is working. I looked up to find how to reset the counter if i open and close the setup menu, the counter is allways on 0. I have to reset all counters to get it work again, but all pictures are destroyed :-(

Can you give me a tip to rest only de printer coutner.

Thankyou

commit 24e63a78 (2019-08-24): Fix #62 by adding max_pages option