pibooth/pibooth #64: How can I change the picture for the states (chosen, print etc) in de language?

thifi, 2019-06-09:
How can I change the picture for the states (chosen, print etc) in de language?
My Photobox has the buttons for taking photo and print on the upper side, so I need the arrows shows up and not down.
I changed/modified the data in /pibooth/pibooth/pictures/de but there are no changes in the pibooth program.
Can you help?

anxuae, 2019-06-09:
Hi Thifi,

You can mask the arrow by setting the config key:
```
[WINDOW]

arrows = False
```

However, if you want the arrows on the top, you have to modify the pictures on the installion path. I guess your modified pictures are in the wrong place.
To find the right location, type the following command:

```
python3 -c "import pibooth;print(pibooth.__file__)"
```

The modified pictures have to be put in ``pibooth/pictures/de`` in this location

anxuae

thifi, 2019-06-09:
Thanks for your comment, Anxuae.
I modified the pictures (with arrows in the other direction)  an copied the files to pibooth/pictures/de, but there is no changing in pibooth.
Is a cache of these files everywhere?
Or have I to delete some files?

Now I made "arrows = False"  but the last picture (made new Photo and you can print...) used a arrow too.

Greetings

thifi, 2019-06-10:
Hi Anxuae,
I have found the problem.
Your command to find the right path shows me the path to python3.5, the update 1.0.4 of pibooth uses the path with python2.7.
I have copied the the remastered images to this path and it works perfectly.
Thanks for your attantion.
Best regards
Thifi

closed 2019-06-10 by thifi