pibooth/pibooth #65: reverse the background png

Alex74300, 2019-06-22:
Hi 
I am trying to reverse the background image (png) and the picture of pibooth. I would like to create a png layer over my picture. I think I need to edit a file .py but I didn't find this line.... Do you have a solution for me?

Thanks 
alex

werdeil, 2019-06-24:
To be sure to understand, you want to have the background png above the pibooth photos instead of below?

This could be a good idea if you manage to have a good png layer

It is not easy to change for now but you can have a look at the pibooth/pictures/concatenate.py, if you find a way feel free to do a pull request

anxuae, 2019-08-26:
First implementation, see: 55082fd9ef698fe246aa5fe28d80d0cdfafb5a10

The key `overlays` permits to define one or two PNG files with transparency areas that will be past over the final picture. Be careful that the overlay shall have the same size (or at least same aspect ratio) than the final picture:
- 2400x3600 for portrait
- 3600x2400 for landscape

closed 2019-09-05 by anxuae

Alex74300, 2019-09-17:
Perfect! Thanks for this enhancement !