pibooth/pibooth #66: Is there a problem with installation at the moment?

deha89, 2019-06-25:
### I'm getting following output after installing all dependencies (see Readme):
`pi@raspberrypi:~ $ sudo pip3 install pibooth`
`Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pibooth in /usr/local/lib/python3.7/dist-packages (1.0.4)
Requirement already satisfied: gphoto2 in /usr/local/lib/python3.7/dist-packages (from pibooth) (2.0.0)
Requirement already satisfied: picamera in /usr/lib/python3/dist-packages (from pibooth) (1.13)
Collecting Pygame==1.9.4 (from pibooth)
  Using cached https://files.pythonhosted.org/packages/b2/6b/c510f0853765eb2219ca5aa3d416d65bb0dea7cd9bb2984aea0a0e04c24d/pygame-1.9.4.tar.gz
    Complete output from command python setup.py egg_info:`
    
    
    WARNING, No "Setup" File Exists, Running "config.py"
    Using UNIX configuration...
    
    
    Hunting dependencies...
    SDL     : found 1.2.15
    FONT    : not found
    IMAGE   : not found
    MIXER   : not found
    PNG     : found
    JPEG    : found
    SCRAP   : found
    PORTMIDI: not found
    PORTTIME: not found
    FREETYPE: found 22.1.16
    Missing dependencies
    
    ----------------------------------------
`Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-8a9q93k4/Pygame/`

Half a year ago, I could install it without any problems. 
Any ideas?
Thanks a lot!
PS It´s a great project.

anxuae, 2019-06-26:
Hi,

This is due to the integration of the lib ``pygame-menu`` as dependancies, it' require pygame >= 1.9.4. I'm working on it to fix this issue.

A workaround, you can install the pygame dependencies by typing this command:

```
sudo apt-get install python-dev libsdl-image1.2-dev libsdl-mixer1.2-dev \
        libsdl-ttf2.0-dev libsdl1.2-dev python-numpy libportmidi-dev libjpeg-dev \
        libtiff5-dev libx11-6 libx11-dev xfonts-base xfonts-100dpi xfonts-75dpi \
        xfonts-cyrillic fluid-soundfont-gm musescore-soundfont-gm
```

https://github.com/bennuttall/piwheels/issues/52

deha89, 2019-06-26:
On my Pi the Problem was the cached pygame version, I guess. After downloading the sources and installing like described on https://www.pygame.org/wiki/GettingStarted with a little edit: `LOCALBASE=/usr/bin python3 setup.py install`.
After that the normal pibooth installation via pip3 worked :-) 
Thanks a lot and greetins to france :-)

closed 2019-06-26 by deha89