pibooth/pibooth #67: Update install Instructions for Raspian Buster

iditude, 2019-06-29:
Raspian buster had been released. From a fresh install, I did have to add a number of packets to get pibooth running. 

Can you maybe update the documentation?

I had to do that before installing pibooth
```
sudo apt install libsdl1.2-dev
sudo apt-get install python-dev libsdl-image1.2-dev libsdl-mixer1.2-dev libsdl-ttf2.0-dev   libsdl1.2-dev libsmpeg-dev python-numpy subversion libportmidi-dev ffmpeg libswscale-dev libavformat-dev libavcodec-dev
sudo apt-get install python-opencv
```

And the opencv line from pip3 doesn't work on that version of raspian. 

I hope this helps future people.

anxuae, 2019-06-29:
Hi Iditude,

Thanks for your feedback, I don't try the buster version yet.
We will update the documentation according to your feedback.

BR
anxuae

Exitras, 2019-07-12:
Hi iditude, thanks a lot for this hint. It also works for me on a Raspberry Pi 4 (4BG RAM). 
However, when I start pibooth, I get the following message:

"File "/usr/local/lib/pzthon3.7/dist-packages/pibooth/controls/light.py", line 65, in __init__
GPIO.setup(pin, GPIO.OUT)
ValueError: The channel sent is invalid on a Raspberry Pi".

Does any of you have an idea of how to go around this issue? As you can imagine, I am pretty new to Python and Raspberry Pi.

iditude, 2019-07-13:
Hello,

Sorry but no idea. I assume Raspberry PI 4 being brand new device, there may be a few things that need to be adjusted for GPIO (pins) to work. Pin numbering may have change.

@anxuae any ideas?

anxuae, 2019-07-16:
Hi Exitras,

Did you change something in the configuration?
I never tested the Pi4, but the GPIO seem to be the same as the Pi3, except some pins have more functionalities (see https://www.tomshardware.com/reviews/raspberry-pi-gpio-pinout,6122.html)

anxuae

Exitras, 2019-07-17:
Hi both, 
many thanks for your reply! I was only installing the packages according to your instructions and the additions by iditude from this post. I´ll try to further investigate this topic and let you know as soon as I can find something out. It would be fantastic if Pibooth would work with a Raspberry Pi4 to build a fast and stable photobooth.

Exitras, 2019-07-17:
I found a way to avoid this error.
Type the following command in the terminal:
cd usr/local/lib/python3.7/dist-packages/pibooth/
sudo chmod 777 booth.py

Open booth.py in usr/local/lib/python3.7/dist-packages/pibooth/
Change to GPIO.setmode(GPIO.BCM) (instead of GPIO.setmode(GPIO.BOARD))

anxuae, 2019-07-18:
> Type the following command in the terminal:
cd usr/local/lib/python3.7/dist-packages/pibooth/
sudo chmod 777 booth.py

This

anxuae, 2019-08-06:
HI @iditude ,

I have updated the installation steps for `Buster with desktop and recommended software`.
However I did not need to execute the commands:
```
sudo apt install libsdl1.2-dev
sudo apt-get install python-dev libsdl-image1.2-dev libsdl-mixer1.2-dev libsdl-ttf2.0-dev   libsdl1.2-dev libsmpeg-dev python-numpy subversion libportmidi-dev ffmpeg libswscale-dev libavformat-dev libavcodec-dev
```

Do you have installed `Raspbian Buster with desktop` or `Raspbian Buster Lite` instead of `Buster with desktop and recommended software` ?

anxuae, 2019-08-07:
Fixed on v1.1.0

closed 2019-08-07 by anxuae