pibooth/pibooth #69: oops, something went wrong

Adridot, 2019-07-16:
I've set everything up, but I don't have a picamera, I only used my nikon D90, with gphoto2.
By using the gphoto commands, I do manage to take pictures via my raspberry pi, however, as soon as I am using pibooth, I am getting this "something went wrong" message, and "[-1] Unexpected error" in the console.
I don't understand why, do you have any solution ?

Exitras, 2019-07-17:
I had the same problem with my Nikon D3100 and found help with this:
https://askubuntu.com/questions/993876/gphoto2-could-not-claim-the-usb-device

Type the following code in your terminal: 
pgrep -fla gphoto2
pkill -f gphoto2

Hope this helps :)

anxuae, 2019-07-20:
Hi @Adridot,

Does the answer of @Exitras helps?

Thanks for your feedback.

anxuae, 2019-08-01:
I'm performed the test on Raspian Buster,
This is mandatory to kill all gphoto2 processes before starting pibooth.

```
pkill -f gphoto2
```

commit 5b307b5a (2019-08-01): Update doc to fix #69 about claiming device interface
closed 2019-08-01 by anxuae