pibooth/pibooth #70: [ ERROR   ] pibooth           : Unsupported setting imgsettings/iso=100

vasi26ro, 2019-07-19:
# Issue:
When I try to take a photo with pibooth I receive this error:
``` bash
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Starting the photo booth application...
[ INFO    ] pibooth           : Connected to printer 'Brother_HL-5250DN_series_via_VNC_from_DESKTOP-LB3RGFG'
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (1 pictures selected)
[ INFO    ] pibooth           : Start new pictures sequence
[ INFO    ] pibooth           : Take picture and save it in /home/pi/Pictures/pibooth/raw/2019-07-19-19-32-50/pibooth000.jpg
[ ERROR   ] pibooth           : Unsupported setting imgsettings/iso=100
```

* I have tried with camera in both PTP and Normal mode same error;
* In PTP mode _"busy"_ is blinking on camera screen;
* `gphoto2 --capture-image` is working.

# Configuration
RPI 3B+ with Raspbian Buster, python 3.7, pygame 1.9.6
Cannon EOS 20D

### Somebody else encountered same problem?

deha89, 2019-07-19:
Hi, not every camera does support to adjust the iso Parameter via p2p. I guess with gphoto2 --list-all-config you will see the possible settings.

vasi26ro, 2019-07-19:
I have tried to comment the iso line from config file but with success because pibooth goes to default config if iso is missing.

Output of `gphoto2 --list-all-config`:

``` bash

Detected a 'Canon:EOS 20D (normal mode)'.                                      
/main/settings/ownername
Label: Owner Name
Readonly: 0
Type: TEXT
Current: Mouse
END
/main/settings/capturesizeclass
Label: Capture Size Class
Readonly: 0
Type: MENU
Current: Compatibility Mode
Choice: 0 Compatibility Mode
Choice: 1 Thumbnail
Choice: 2 Full Image
END
/main/settings/iso
Label: ISO Speed
Readonly: 0
Type: MENU
Current: 400
Choice: 0 50
Choice: 1 100
Choice: 2 125
Choice: 3 160
Choice: 4 200
Choice: 5 250
Choice: 6 320
Choice: 7 400
Choice: 8 500
Choice: 9 640
Choice: 10 800
Choice: 11 1000
Choice: 12 1250
Choice: 13 1600
Choice: 14 3200
END
/main/settings/shootingmode
Label: Shooting mode
Readonly: 0
Type: MENU
Current: AUTO
Choice: 0 AUTO
Choice: 1 P
Choice: 2 Tv
Choice: 3 Av
Choice: 4 M
Choice: 5 A-DEP
Choice: 6 M-DEP
Choice: 7 Bulb
Choice: 8 Manual 2
Choice: 9 Far scene
Choice: 10 Fast shutter
Choice: 11 Slow shutter
Choice: 12 Night scene
Choice: 13 Gray scale
Choice: 14 Sepia
Choice: 15 Portrait
Choice: 16 Spot
Choice: 17 Macro
Choice: 18 BW
Choice: 19 PanFocus
Choice: 20 Vivid
Choice: 21 Neutral
Choice: 22 Flash off
Choice: 23 Long shutter
Choice: 24 Super macro
Choice: 25 Foliage
Choice: 26 Indoor
Choice: 27 Fireworks
Choice: 28 Beach
Choice: 29 Underwater
Choice: 30 Snow
Choice: 31 Kids and pets
Choice: 32 Night snapshot
Choice: 33 Digital macro
Choice: 34 MyColors
Choice: 35 Photo in movie
END
/main/settings/shutterspeed
Label: Shutter Speed
Readonly: 0
Type: MENU
Current: 1
Choice: 0 Bulb
Choice: 1 30
Choice: 2 25
Choice: 3 20
Choice: 4 15
Choice: 5 13
Choice: 6 10
Choice: 7 8
Choice: 8 6
Choice: 9 5
Choice: 10 4
Choice: 11 3.2
Choice: 12 2.5
Choice: 13 2
Choice: 14 1.6
Choice: 15 1.3
Choice: 16 1
Choice: 17 0.8
Choice: 18 0.6
Choice: 19 0.5
Choice: 20 0.4
Choice: 21 0.3
Choice: 22 1/4
Choice: 23 1/5
Choice: 24 1/6
Choice: 25 1/8
Choice: 26 1/10
Choice: 27 1/13
Choice: 28 1/15
Choice: 29 1/20
Choice: 30 1/25
Choice: 31 1/30
Choice: 32 1/40
Choice: 33 1/50
Choice: 34 1/60
Choice: 35 1/80
Choice: 36 1/100
Choice: 37 1/125
Choice: 38 1/160
Choice: 39 1/200
Choice: 40 1/250
Choice: 41 1/320
Choice: 42 1/400
Choice: 43 1/500
Choice: 44 1/640
Choice: 45 1/800
Choice: 46 1/1000
Choice: 47 1/1250
Choice: 48 1/1600
Choice: 49 1/2000
Choice: 50 1/2500
Choice: 51 1/3200
Choice: 52 1/4000
Choice: 53 1/5000
Choice: 54 1/6400
Choice: 55 1/8000
END
/main/settings/aperture
Label: Aperture
Readonly: 0
Type: MENU
Current: 5.6
Choice: 0 1.2
Choice: 1 1.4
Choice: 2 1.6
Choice: 3 1.8
Choice: 4 2.0
Choice: 5 2.2
Choice: 6 2.5
Choice: 7 2.8
Choice: 8 3.2
Choice: 9 3.5
Choice: 10 4.0
Choice: 11 4.5
Choice: 12 5.0
Choice: 13 5.6
Choice: 14 6.3
Choice: 15 7.1
Choice: 16 8
Choice: 17 9
Choice: 18 10
Choice: 19 11
Choice: 20 13
Choice: 21 14
Choice: 22 16
Choice: 23 18
Choice: 24 20
Choice: 25 22
Choice: 26 25
Choice: 27 29
Choice: 28 32
END
/main/settings/exposurecompensation
Label: Exposure Compensation
Readonly: 0
Type: MENU
Current: 0
Choice: 0 +2
Choice: 1 +1 2/3
Choice: 2 +1 1/2
Choice: 3 +1 1/3
Choice: 4 +1
Choice: 5 +2/3
Choice: 6 +1/2
Choice: 7 +1/3
Choice: 8 0
Choice: 9 -1/3
Choice: 10 -1/2
Choice: 11 -2/3
Choice: 12 -1
Choice: 13 -1 1/3
Choice: 14 -1 1/2
Choice: 15 -1 2/3
Choice: 16 -2
END
/main/settings/imageformat
Label: Image Format
Readonly: 0
Type: MENU
Current: Medium Fine JPEG
Choice: 0 RAW
Choice: 1 RAW 2
Choice: 2 Small Normal JPEG
Choice: 3 Small Fine JPEG
Choice: 4 Medium Normal JPEG
Choice: 5 Medium Fine JPEG
Choice: 6 Large Normal JPEG
Choice: 7 Large Fine JPEG
Choice: 8 RAW + Small Normal JPEG
Choice: 9 RAW + Small Fine JPEG
Choice: 10 RAW + Medium Normal JPEG
Choice: 11 RAW + Medium Fine JPEG
Choice: 12 RAW + Large Normal JPEG
Choice: 13 RAW + Large Fine JPEG
END
/main/settings/focusmode
Label: Focus Mode
Readonly: 0
Type: MENU
Current: Auto focus: AI focus
Choice: 0 Auto focus: one-shot
Choice: 1 Auto focus: AI servo
Choice: 2 Auto focus: AI focus
Choice: 3 Manual focus
END
/main/settings/flashmode
Label: Flash Mode
Readonly: 0
Type: MENU
Current: Unknown
Choice: 0 Flash off
Choice: 1 Flash on
Choice: 2 Flash auto
Choice: 3 Unknown
END
/main/settings/beep
Label: Beep
Readonly: 0
Type: MENU
Current: Beep on
Choice: 0 Beep off
Choice: 1 Beep on
END
/main/actions/syncdatetime
Label: Synchronize camera date and time with PC
Readonly: 0
Type: TOGGLE
Current: 0
END
/main/status/model
Label: Camera Model
Readonly: 0
Type: TEXT
Current: Canon EOS 20D
END
/main/status/datetime
Label: Date and Time
Readonly: 0
Type: TEXT
Current: 2019-07-19 20:21:05
END
/main/status/firmwareversion
Label: Firmware Version
Readonly: 0
Type: TEXT
Current: 2.0.3.0
END
/main/status/driver
Label: Power
Readonly: 0
Type: TEXT
Current: on battery (power OK)
END
/main/Driver/list_all_files
Label: List all files
Readonly: 0
Type: TOGGLE
Current: 0
END
```

deha89, 2019-07-19:
You could download the developer version for testing and remove the iso part from camera.py (begins in line 180 I think). Or someone has another idea.

anxuae, 2019-07-19:
The error is due to gphoto2 config keys are dependent of the camera brand, thus it is difficult
 to make a generic code.

The only solution that I see, is to define config key in the `pibooth` config file.

*Note:* be careful, the current version if not fully functional.

anxuae, 2019-07-21:
Setting parameters which are not supported don't raise exception (ba37ca2)

closed 2019-07-21 by anxuae

vasi26ro, 2019-07-22:
@anxuae  Setting ISO to 100 manually on the DSLR  camera didn't solve the problem.

vasi26ro, 2019-07-22:
In my opinion the problem is with this so called "endpoints":

``` bash

Detected a 'Canon:EOS 20D (normal mode)'.                                      
/main/settings/ownername
/main/settings/capturesizeclass
/main/settings/iso
/main/settings/shootingmode
/main/settings/shutterspeed
/main/settings/aperture
/main/settings/exposurecompensation
/main/settings/imageformat
/main/settings/focusmode
/main/settings/flashmode
/main/settings/beep
/main/actions/syncdatetime
/main/status/model
/main/status/datetime
/main/status/firmwareversion
/main/status/driver
/main/Driver/list_all_files

```

Pibooth is looking for `imgsettings/iso=100` and for this Canon EOS 20D this setting is available at: `/main/settings/iso`

anxuae, 2019-07-22:
Hi vasi26ro,

Yes you are right, that's why I don't raise exception any more, but only display an ERROR message in the console logs.
My changes will be available in the next version of `pibooth` which is not yet released.

anxuae

vasi26ro, 2019-07-22:
Thank you @anxuae, great project!

themaluxis, 2019-07-30:
I've this error : 
[ ERROR   ] pibooth           : Unsupported setting imgsettings/iso=100

And here is the result of the --list-all-config command : 

> /main/imgsettings/iso
Label: ISO Speed
Readonly: 0
Type: RADIO
Current: 100
Choice: 0 Auto
Choice: 1 100
Choice: 2 200
Choice: 3 400
Choice: 4 800
Choice: 5 1600
Choice: 6 3200
Choice: 7 6400
Choice: 8 12800
Choice: 9 25600