pibooth/pibooth #72: [ ERROR ] pibooth : Unsupported setting imgsettings/iso=100

themaluxis, 2019-07-31:
I've this error :
[ ERROR ] pibooth : Unsupported setting imgsettings/iso=100

And here is the result of the --list-all-config command :

/main/imgsettings/iso
Label: ISO Speed
Readonly: 0
Type: RADIO
Current: 100
Choice: 0 Auto
Choice: 1 100
Choice: 2 200
Choice: 3 400
Choice: 4 800
Choice: 5 1600
Choice: 6 3200
Choice: 7 6400
Choice: 8 12800
Choice: 9 25600

What Can I DO pls ?

anxuae, 2019-07-31:
Hi @themaluxis 

Just wait for the new release of `pibooth` next week.
(alternatively, you can install the code of the currently developed version, see https://github.com/werdeil/pibooth/blob/master/docs/dev.rst, be careful, the code may be unstable)

BR

themaluxis, 2019-07-31:
[ ERROR   ] pibooth           : cannot identify image file <_io.BytesIO object at 0x717f6540>
with this method :-/

themaluxis, 2019-07-31:
verbose output : 

> [ INFO    ] pibooth           : +< Take picture and save it in /home/pi/Pictures/pibooth/raw/2019-07-31-13-48-41/pibooth000.jpg
[ DEBUG   ] pibooth           : |  Setting option actions/viewfinder=2
[ DEBUG   ] pibooth           : +> took 5.971 seconds
[ DEBUG   ] pibooth           : Activate state 'processing'
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IHDR' 16 13
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'iCCP' 41 389
[ DEBUG   ] PIL.PngImagePlugin: iCCP profile name b'ICC profile'
[ DEBUG   ] PIL.PngImagePlugin: Compression method 0
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'bKGD' 442 6
[ DEBUG   ] PIL.PngImagePlugin: b'bKGD' 442 6 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'pHYs' 460 9
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tIME' 481 7
[ DEBUG   ] PIL.PngImagePlugin: b'tIME' 481 7 (unknown)
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'tEXt' 500 25
[ DEBUG   ] PIL.PngImagePlugin: STREAM b'IDAT' 537 8192
[ INFO    ] pibooth           : +< Creating merged picture
[ DEBUG   ] PIL.Image         : |  Importing BmpImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing BufrStubImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing CurImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing DcxImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing DdsImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing EpsImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing FitsStubImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing FliImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing FpxImagePlugin
[ DEBUG   ] PIL.Image         : |  Image: failed to import FpxImagePlugin: No module named 'olefile'
[ DEBUG   ] PIL.Image         : |  Importing FtexImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing GbrImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing GifImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing GribStubImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing Hdf5StubImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing IcnsImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing IcoImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing ImImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing ImtImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing IptcImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing JpegImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing Jpeg2KImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing McIdasImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing MicImagePlugin
[ DEBUG   ] PIL.Image         : |  Image: failed to import MicImagePlugin: No module named 'olefile'
[ DEBUG   ] PIL.Image         : |  Importing MpegImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing MpoImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing MspImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing PalmImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing PcdImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing PcxImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing PdfImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing PixarImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing PngImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing PpmImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing PsdImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing SgiImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing SpiderImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing SunImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing TgaImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing TiffImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing WebPImagePlugin
[ DEBUG   ] PIL.Image         : |  Image: failed to import WebPImagePlugin: cannot import name '_webp'
[ DEBUG   ] PIL.Image         : |  Importing WmfImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing XbmImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing XpmImagePlugin
[ DEBUG   ] PIL.Image         : |  Importing XVThumbImagePlugin
[ DEBUG   ] pibooth           : +> took 1.920 seconds
[ ERROR   ] pibooth           : cannot identify image file <_io.BytesIO object at 0x717f5720>
Traceback (most recent call last):
  File "/home/pi/pibooth/pibooth/states.py", line 75, in process
    self.active_state.do_actions(events)
  File "/home/pi/pibooth/pibooth/booth.py", line 259, in do_actions
    self.app.previous_picture = concatenate_pictures(self.app.camera.get_captures(),
  File "/home/pi/pibooth/pibooth/controls/camera/base.py", line 79, in get_captures
    images.append(self._post_process_capture(path))
  File "/home/pi/pibooth/pibooth/controls/camera/gphoto.py", line 118, in _post_process_capture
    image = Image.open(io.BytesIO(camera_file.get_data_and_size()))
  File "/usr/local/lib/python3.5/dist-packages/PIL/Image.py", line 2585, in open
    % (filename if filename else fp))
OSError: cannot identify image file <_io.BytesIO object at 0x717f5720>
[ DEBUG   ] pibooth           : Activate state 'failsafe'

I'm gonna try a fresh isntall

anxuae, 2019-07-31:
Ok, the error is not the same.
Can you tell me the which picture format do have you choose in your DSLR camera?

Best is to paste the output of the command `gphoto2 --list-all-config`.

themaluxis, 2019-08-01:
Here is for you 
`/main/actions/uilock
Label: UI Lock
Type: TOGGLE
Current: 2
/main/actions/autofocusdrive
Label: Drive Canon DSLR Autofocus
Type: TOGGLE
Current: 0
/main/actions/manualfocusdrive
Label: Drive Canon DSLR Manual focus
Type: RADIO
Current: None
Choice: 0 Near 1
Choice: 1 Near 2
Choice: 2 Near 3
Choice: 3 None
Choice: 4 Far 1
Choice: 5 Far 2
Choice: 6 Far 3
/main/actions/cancelautofocus
Label: Cancel Canon DSLR Autofocus
Type: TOGGLE
Current: 0
/main/actions/eoszoom
Label: Canon EOS Zoom
Type: TEXT
Current: 0
/main/actions/eoszoomposition
Label: Canon EOS Zoom Position
Type: TEXT
Current: 0,0
/main/actions/viewfinder
Label: Canon EOS Viewfinder
Type: TOGGLE
Current: 2
/main/actions/eosremoterelease
Label: Canon EOS Remote Release
Type: RADIO
Current: None
Choice: 0 None
Choice: 1 Press Half
Choice: 2 Press Full
Choice: 3 Release Half
Choice: 4 Release Full
Choice: 5 Immediate
Choice: 6 Press 1
Choice: 7 Press 2
Choice: 8 Press 3
Choice: 9 Release 1
Choice: 10 Release 2
Choice: 11 Release 3
/main/actions/opcode
Label: PTP Opcode
Type: TEXT
Current: 0x1001,0xparam1,0xparam2
/main/settings/capturetarget
Label: Capture Target
Type: RADIO
Current: Internal RAM
Choice: 0 Internal RAM
Choice: 1 Memory card
/main/settings/capture
Label: Capture
Type: TOGGLE
Current: 0
/main/status/serialnumber
Label: Serial Number
Type: TEXT
Current: 488dabb039314c339a3b752c84a7af96
/main/status/manufacturer
Label: Camera Manufacturer
Type: TEXT
Current: Canon Inc.
/main/status/cameramodel
Label: Camera Model
Type: TEXT
Current: Canon EOS 750D
/main/status/deviceversion
Label: Device Version
Type: TEXT
Current: 3-1.0.0
/main/status/vendorextension
Label: Vendor Extension
Type: TEXT
Current: None
/main/status/Battery Level
Label: Battery Level
Type: TEXT
Current: 100%
/main/other/d049
Label: Model ID
Type: TEXT
Current: -2147482733
/main/other/d402
Label: PTP Property 0xd402
Type: TEXT
Current: Canon EOS 750D
/main/other/d407
Label: PTP Property 0xd407
Type: TEXT
Current: 1
/main/other/d406
Label: PTP Property 0xd406
Type: TEXT
Current: Unknown Initiator
/main/other/d303
Label: PTP Property 0xd303
Type: TEXT
Current: 1
/main/other/5001
Label: Battery Level
Type: MENU
Current: 100
Choice: 0 100
Choice: 1 0
Choice: 2 75
Choice: 3 0
Choice: 4 50
`

anxuae, 2019-08-01:
Thanks, I can not see the property `imageformat` in your log.
According to your error, I'm guessing that the file forma, provided by your camera, is not recognized by Pillow (image library). Can you double-check that you didn't choose `RAW` or `RAW+ something else` ?

For instance on my `Canon EOS 1100D`, I ave the following settings, I have manually choose the `Medium Fine JPEG` format:

```
/main/imgsettings/imageformat
Label: Image Format
Readonly: 0
Type: RADIO
Current: Medium Fine JPEG
Choice: 0 Large Fine JPEG
Choice: 1 Large Normal JPEG
Choice: 2 Medium Fine JPEG
Choice: 3 Medium Normal JPEG
Choice: 4 Small Fine JPEG
Choice: 5 Small Normal JPEG
Choice: 6 Smaller JPEG
Choice: 7 Tiny JPEG
Choice: 8 RAW + Large Fine JPEG
Choice: 9 RAW
```

anxuae, 2019-08-07:
Fixed on v1.1.0

closed 2019-08-07 by anxuae