pibooth/pibooth #76: No camera detected

Doskon, 2019-08-07:
Hi,
I have installed pibooth but I get the error "Neither PiCamera nor GPhoto2 camera detected" when I try to start pibooth. I am using the Canon EOS 1300D and with GPhoto2 I am able to take photos directly. 

This is the complete error message: 
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 653, in main
    app = PiApplication(config)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 375, in __init__
    raise EnvironmentError("Neither PiCamera nor GPhoto2 camera detected")

Thanks for help!

anxuae, 2019-08-07:
Hi @Doskon ,

Can you clone this project an execute the diagnostic script? Here are the steps to execute in a shell:

```
git clone https://github.com/werdeil/pibooth.git
cd pibooth
python3 tests/dslr-diagnostic
```

Doskon, 2019-08-08:
[dslr-diagnostic.log](https://github.com/werdeil/pibooth/files/3483691/dslr-diagnostic.log)

anxuae, 2019-08-09:
Huhu strange, the diagnosis script is working well, thus it mean that the camera can be driven by the pibooth code.

Really I have no idea where the problem comes from. I made changes in the diag script.

Can you run it again?

Doskon, 2019-08-11:
[dslr-diagnostic.log](https://github.com/werdeil/pibooth/files/3489615/dslr-diagnostic.log)

anxuae, 2019-08-16:
Hi,

Can you do the following actions, and tell me is its working?

1. Uninstall pibooth:
    ```
    sudo pip3 uninstall pibooth
    ```
2. in your cloned repository, modify the file `pibooth/booth.py` by replacing the following lines
https://github.com/werdeil/pibooth/blob/64e7550382a4d7a6e54f81a6bd2f3a38d871a9b2/pibooth/booth.py#L368-L375
with
    ```python
    cam_class = camera.GpCamera
    ```
3. Run pibooth from cloned repository, run the command in the repository:
    ```
    export PYTHONPATH=$(pwd);python3 pibooth/booth.py
    ```

anxuae, 2019-08-24:
Hi, do you have some news about this topic?

bennnil, 2019-09-01:
I am having the same problem. 
When I change your code it works like charm, however the preview updates very slow. When I quit pibooth and restart it, I have to switch the camera off and on again so that it works.

Is there a way to get the Hybrid Mode for me?
PS: Here is my log file: [dslr-diagnostic.log](https://github.com/werdeil/pibooth/files/3563748/dslr-diagnostic.log)

anxuae, 2019-09-03:
Hi @bennnil ,

1. When you say
  > When I change your code

  would you mean changed as explained in my previous post? Sorry for this foolish question but I can't reproduce this issue and I have no idea how to investigate. The `dslr-diagnostic` script is working find and it uses exactly the same commands that `pibooth`, thus it can not be a help for me.

2. With only a DSLR camera connected, update is slow because each frame is taken one by one and resized to fit the desired resolution and the screen. I'm currently trying to optimize that but it will never be smooth.

3. To get the Hybrid mode, you need your DSLR camera connected and a Raspberry Pi camera (the preview is taken using the Pi camera).

anxuae

bennnil, 2019-09-04:
Hey @anxuae ,
thanks for your fast reply. 
I just checked out again the current code from github and tested without the change as explained in your previous post. Now the Hybrid mode works fine. I guess the problem was, that I had to restart the camera (switch off and the on again) everytime I restart the program. I do not know why but otherwise the camera is not detected.
I guess restarting the camera might have helped @Doskon as well. So for me the "bug" is resolved. Thanks for your help!

anxuae, 2019-09-04:
Hi @bennnil ,

Can you give me the error that is displayed if you don't turn off/on your camera?
(Set the param `failsafe = False` in the config before).

Thanks

bennnil, 2019-09-04:
Actually if I don't connect the camera after booting but before it is not even detected says the DSLR diagnostics. 

Pibooth does not throw any errors, instead the picamera is used for both preview and capturing. 
If I then restart the camera everything works. 
In both cases there is nothing on the display of my camera shown.

werdeil, 2019-09-04:
Seems that your DSLR is not mounted/connected if it is started before the pi, there is maybe a setting on the DSLR that put it in a "battery save mode" that prevents USB communication.

anxuae, 2019-09-13:
@Doskon can we close this issue or you're still facing to the problem ?

anxuae, 2019-09-22:
No news, I close this issue.
Please open a new issue if the problem is still present.

closed 2019-09-22 by anxuae