pibooth/pibooth #80: pibooth won't start - pygame.error: video system not initialized

dsheard, 2019-09-01:
Getting the following issue when running pibooth. Have wiped my SD and started from scratch but still the same issue.

```
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Starting the photo booth application...
Traceback (most recent call last):
  File "/usr/local/bin/pibooth", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 711, in main
    app = PiApplication(config)
  File "/usr/local/lib/python3.7/dist-packages/pibooth/booth.py", line 392, in __init__
    pygame.event.set_blocked(pygame.MOUSEMOTION)
pygame.error: video system not initialized
```

dsheard, 2019-09-01:
Camera was not working! Fixed

dsheard, 2019-09-01:
Correction: still does this when running over ssh. Have to open VNC to get it working.

closed 2019-09-01 by dsheard

anxuae, 2019-09-01:
Hi @dsheard,

I think that pygame display can not be started over ssh. You need a display plugged to your Raspberry Pi.

anxuae