pibooth/pibooth #82: Why oh why is the default 'clear_on_startup=True'?

etiennedm, 2019-09-16:
This happened this weekend: the booth started to show an error and I realized that the SD card of the raspberry was full. So far so good, it's what it should do. I stopped pibooth, plugged a USB flashdrive and started to edit the config file to point the storage location of the new pictures to the flash drive. 

However for some reason when I saved the config it was simply deleted (and I suspect this has something to do with the SD card being full and me trying to write an updated file). Except I did not realize this right away, instead I just restarted pibooth to take the change into account. When I saw that the footer text was back to its default value, I went to edit the config again but it was gone. And when I checked the pictures folder it was now empty since the default value of 'clear_on_startup' is True.

Should we maybe make the default False? Any good reason to keep it to True since it is extremely dangerous during normal use of the booth?

Other than that it worked flawlessly, and I just realized that the pictures should be on the camera memory card anyway. I just won't have the montages.

werdeil, 2019-09-16:
I agree with @etiennedm, I'll change the default value.

werdeil, 2019-09-16:
Fix by 5c49b6be186c82fe1c686d93f3dbc4a8ec6291c5

closed 2019-09-16 by werdeil