pibooth/pibooth #83: disable pibooth menu config

Alex74300, 2019-09-17:
**Describe the bug**
A clear and concise description of what the bug is.

**Steps to reproduce the bug**
1. Go to pibooth, take a picture with button GPIO1 and GPIO2, print a picture avec GPIO button
2. Click on button GPIO1 or GPIO2 to become a session picture and the pibooth menu config  display on screen on the place of choose screen.

I think the button GPio 1 and 2 are 2 functions at the second time of pibooth.

What's the procedure to disable this screen menu config?

Thanks .

alex

**Environment**
 - OS: [RPI linux gnome]
 - Camera Rpi camera
 - Pibooth version [e.g. 1.1.2]

anxuae, 2019-09-17:
Hi @Alex74300,

Can you try by setting a bigger 'debounce_delay' in the config? By default it is 0.5, maybe try with 0.8.

anxuae

anxuae, 2019-09-22:
Hi @Alex74300 ,

If changing the `debounce_delay` does not solves the problem, you can comment the following line in the code:

https://github.com/werdeil/pibooth/blob/e54a8f1dd494fadcacfaa33071aafda9ce2ecb73/pibooth/booth.py#L632

anxuae

anxuae, 2019-10-02:
Is this solution solve your problem?
If yes, please close the bug.

Alex74300, 2019-10-18:
Thanks !! Yes this solution solve my bug ;)