pibooth/pibooth #84: Ca

andhey, 2019-09-20:
Hello you,
i have installed pibooth. All functions as expected, only the photos are out of focus. Without pibooth, the camera works normally.
I press the image button, the preview is displayed blurred, the Kamara triggers and then a blurred photo is displayed on the monitor. The camera does not focus on the picture before recording.

**Logs**
1. config file:
[pibooth.txt](https://github.com/werdeil/pibooth/files/3637595/pibooth.txt)
2. logs from the console 
[piboot_v.txt](https://github.com/werdeil/pibooth/files/3637586/piboot_v.txt)
3. dslr-log
[dslr-diagnostic.txt](https://github.com/werdeil/pibooth/files/3637620/dslr-diagnostic.txt)

**Environment**
 - OS:  Raspbian Buster
 - Camera: Canon EOS 100D
 - Pibooth version 1.1.2

werdeil, 2019-09-20:
> /actions/autofocusdrive
  Label       : Drive Canon DSLR Autofocus
  Readonly    : no
  Data type   : <class 'int'>
  Widget type : Toggle button (or check box)
  Current     : 0
  Choices     : [0, 1]

I would say that your camera is in autofocus mode off and we don't set this setting in pibooth. You can try to see if there is a setting in your 1000D for this.

@anxuae What do you think? Shall we try to set this setting?

andhey, 2019-09-21:
Hello 

all autofocus settings in the camera are on. I can not find a suitable function otherwise.
which code would I have to insert at which point, so that the autofocus is controlled by pibooth.

thank you for your efforts
great job

anxuae, 2019-09-22:
Hi @andhey ,

As proposed by @werdeil, we can manage to set the autofocus in the code if it's working. Can you try it? 
1. Uninstall the currently installed version of `pibooth`
2. Install a developing version: https://github.com/werdeil/pibooth/blob/master/docs/dev.rst
3. Add the following line in the cloned repository
    ```
    self.set_config_value('actions', 'autofocusdrive', 1)
    ```
    below lines:
    https://github.com/werdeil/pibooth/blob/e54a8f1dd494fadcacfaa33071aafda9ce2ecb73/pibooth/controls/camera/gphoto.py#L80-L82

If it works, we could add it permanently in the next `pibooth` release.

anxuae

andhey, 2019-09-22:
Hi anxuae,

thanks for your help.  Unfortunately did not work. Images are still not focused. Can I give the event trigger half pressed by gphoto2 to the camera?

andhey

andhey, 2019-09-23:
Hello again,

when searching the web I found the following:
A user of gphoto reported the following:
"[...] It seems if I write the following in command:

gphoto2 --set-config / main / actions / autofocusdrive = 1

it wants focus. However, it does not change focus for the next photo. If I write the following command:

gphoto2 --set-config / main / actions / autofocusdrive = 1 --capture-image-and-download [...] "

I try this tonight only with gphoto2,
How can I do this in pibooth?

anxuae, 2019-09-23:
Hi,

You can try to put the line:

```
self.set_config_value('actions', 'autofocusdrive', 1)
```

just before:

https://github.com/werdeil/pibooth/blob/50ff49a2e9b26a77448343e3682c61dc127c36ac/pibooth/controls/camera/gphoto.py#L251

anxuae

anxuae, 2019-09-23:
Moreover, on our configuration, the `Sports` mode is used.
Be sure also that the button on your lens is toggled to `AF`.

andhey, 2019-09-23:
Hi anxuae,

First of all thanks for your support. By manually controlling the camera via gphoto2, I was able to narrow down the problem.
As soon as the mirror is up, no focusing takes place. This only takes place when the mirror is open. A solution to the problem at least for my EOS 1000D would be the following procedure:
1) Take photo at open mirror and discard the photo
2) start Preview with the mirror up
3) Open the mirror again and take a picture

is that possible to implement?

andhey, 2019-09-23:
I solved the Problem for me. I made two changes in the gphoto.py:

1) insert
`self._cam.capture(gp.GP_CAPTURE_IMAGE), effect`
before
https://github.com/werdeil/pibooth/blob/50ff49a2e9b26a77448343e3682c61dc127c36ac/pibooth/controls/camera/gphoto.py#L175

2) insert
`self.set_config_value('actions', 'viewfinder', 0)`
before
https://github.com/werdeil/pibooth/blob/e54a8f1dd494fadcacfaa33071aafda9ce2ecb73/pibooth/controls/camera/gphoto.py#L251

that's for your support. Would be nice if you can consider that in a later version.
For me, issue is closed.

anxuae, 2019-09-24:
Hi @andhey,

Thanks for your feedback. I have not this problem using my Canon EOS REBEL T3.
However your issue permits to see a bug on the initial value of the parameter `viewfinder`.

BR
anxuae

andhey, 2019-09-24:
Hi @anxuae,
sorry for reopening.
it is correct that the images are not deleted from the SD card. By my change now always two photo files for each pass remain on the SD card. If I cancel my change, I still have one image per pass on the SD card.

anxuae, 2019-09-24:
Yes of course, call to `self._cam.capture` function will take a picture.
Did you try if the focus is correct only by adding `self.set_config_value('actions', 'viewfinder', 0)` before line 251?

My last commit today implement this change.

anxuae, 2019-09-24:
if not, you try to use `self._cam.capture_preview()` intstead of `self._cam.capture(gp.GP_CAPTURE_IMAGE)` on line 175. This method does not save picture on memory card.

andhey, 2019-09-25:
@anxuae 

> Yes of course, call to self._cam.capture function will take a picture.

That was probably a misunderstanding. My question was, whether it is wanted by pibooth that the recordings remain on the SD card and not automatically deleted.

> Did you try if the focus is correct only by adding `self.set_config_value('actions', 'viewfinder', 0)` before line 251?

 In that case the preview is out of focus an the photo is o.k.

> If not, you can try to use self._cam.capture_preview() intstead of self._cam.capture(gp.GP_CAPTURE_IMAGE) before line 175. This method does not save picture on memory card.

In that case the preview also is out of focus an the photo is o.k. 

I ned `self._cam.capture(gp.GP_CAPTURE_IMAGE)` to focus whit the autofocus-function. I did not find another  option.

anxuae, 2019-09-25:
Hi,

>That was probably a misunderstanding. My question was, whether it is wanted by pibooth that the recordings remain on the SD card and not automatically deleted.
Yes it is, it is a kind of backup in case of crash of the raspberry pi. But this could be optional if you don't have enough space on your SD.

anxuae

andhey, 2019-09-25:
@anxuae 
That would be a nice feature for me.

commit dda8fc17 (2019-09-25): Fix #84: add 'delete_internal_memory' option
closed 2019-09-25 by anxuae