pibooth/pibooth #87: raw photo blue

MickaelB, 2019-10-05:
**Describe the bug**
with an opencv webcam (logitech C920) when we take photo.
render photo is ok in pibooth folder, but in raw folder photo are blue

**Steps to reproduce the bug**
1. take photo
2. go to raw folder
3. open photo
4. photo have blue filter

**Logs**
1. Set the `debug` mode in the config
2. Post your config file >> default
3. Copy-paste all the logs from the console
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
[ INFO    ] pibooth           : Starting the photo booth application...
[ INFO    ] pibooth           : Configuring OpenCV camera ...
VIDEOIO ERROR: V4L2: setting property #30 is not supported
[ WARNING ] pibooth           : No printer configured in CUPS (see http://localhost:631)
[ INFO    ] pibooth           : Show picture choice (nothing selected)
[ INFO    ] pibooth           : Show picture choice (4 captures selected)
[ INFO    ] pibooth           : Start new captures sequence
[ INFO    ] pibooth           : Take a capture and save it in /home/pi/Pictures/pibooth/raw/2019-10-05-12-37-52/pibooth000.jpg
[ INFO    ] pibooth           : Take a capture and save it in /home/pi/Pictures/pibooth/raw/2019-10-05-12-37-52/pibooth001.jpg
[ INFO    ] pibooth           : Take a capture and save it in /home/pi/Pictures/pibooth/raw/2019-10-05-12-37-52/pibooth002.jpg
[ INFO    ] pibooth           : Take a capture and save it in /home/pi/Pictures/pibooth/raw/2019-10-05-12-37-52/pibooth003.jpg
[ INFO    ] pibooth           : Creating the final picture
[ INFO    ] pibooth           : Use OpenCvPictureMaker to create background
[ INFO    ] pibooth           : Use OpenCvPictureMaker to concatenate images
[ INFO    ] pibooth           : Use OpenCvPictureMaker to assemble final image
[ INFO    ] pibooth           : Use OpenCvPictureMaker to draw texts
[ INFO    ] pibooth           : Save image '/home/pi/Pictures/pibooth/2019-10-05-12-37-52_pibooth.jpg'
[ INFO    ] pibooth           : Show picture choice (nothing selected)pi@raspber
4. Post the `diagnostic.log` file if you are using a DSLR camera (generated by running the command `pibooth --diagnostic`)

**Environment**
 - OS [Raspbian Buster]
 - Camera [logitech C920]
 - Pibooth version [1.9.4.post1]

many thanks for this great photobooth

anxuae, 2019-10-06:
Hi @MickaelB 

Thanks for your bug report. The fix will be integrated to the `pibooth` 1.1.5 (coming soon).

anxuae

closed 2019-10-06 by anxuae

sblanc27, 2019-11-05:
Hi, 
Thanks for your nice application.
May I ask when will come the version 1.1.5 which fixes this issue ? I would require quite soon.
Tx

commit 46b4d249 (2019-10-06): Fix #87: convert image to BGR before save it in CvCamera