pibooth/pibooth #90: Help us to add new languages!!

werdeil, 2019-10-18:
We've recently implemented a better way to handle languages in pibooth and it is now easy to add new ones.

If you want pibooth to support your language, please send us the translation of the following texts in the same format:
```
    'en': {
        'intro': "Take a photo",
        'intro_print': 'Or you can\nstill print\nthis photo',
        'choose': "Choose your layout",
        '1': "1 photo",
        '2': "2 photos",
        '3': "3 photos",
        '4': "4 photos",
        'chosen': "Let's go!",
        'smile': "Smile !",
        'processing': "Processing...",
        'print': "Print the photo?",
        'finished': "Thanks",
        'oops': "Oops something went wrong",
    },
```

Thanks a lot!

nihuel, 2019-10-23:
Hi.. I've generated language Spanish for pibooth
The translation's name is 'es-ar' because I'm from Argentina, and there are some slight differences with the one spoken in Spain. 

<pre>
<code>
'es-ar': {
       'intro': "Sacate Una Foto",
       'intro_print': 'O Podés\nTodavia Imprimir\nEsta Foto',
       'choose': "Elegí el tipo De Foto",
       '1': "1 Foto",
       '2': "2 Fotos",
       '3': "3 Fotos",
       '4': "4 Fotos",
       'chosen': "Adelante!!",
       'smile': "Sonríe!!",
       'processing': "Procesando...",
       'print': "Imprimimos esta foto?",
       'finished': "Gracias",
       'oops': "D'oh! Algo salio mal 0_o",
    }
</code>
</pre>

I hope you find it useful.
Thanks..

Real2222, 2019-10-23:
'O Podés\nTodavia Imprimir\nEsta Foto'

Are you sure ?
I dont think that "podés" are the right translation, this word doesnt exist.

werdeil, 2019-10-24:
Thanks a lot @nihuel and @Real2222! We will be glad to add Spanish to the supported languages.

However I'm not sure we want to introduce the region part for the translation, can we agree on a Spanish version which is understood in all regions?

'es': {
       'intro': "Sacate Una Foto",
       'intro_print': 'O Puedes\nTodavia Imprimir\nEsta Foto',
       'choose': "Elege el tipo de foto",
       '1': "1 Foto",
       '2': "2 Fotos",
       '3': "3 Fotos",
       '4': "4 Fotos",
       'chosen': "Adelante!!",
       'smile': "Sonríe!!",
       'processing': "Procesando...",
       'print': "Imprimir esta foto?",
       'finished': "Gracias",
       'oops': "Maldición! Algo salió mal",
    }

nihuel, 2019-10-24:
@Real2222
It is the right translation if, like I said on my post, 
you are from Argentina or Uruguay where 'vos' is most commonly used instead of 'tú'. 
So ‘Podés’ is the second person singular of the indicative form in the present.

@werdeil
Yes, no problem, I undestand, it is ok by me. 
only one thing.  on 'choose': "Elege el tipo de foto". 
Is "Elige" instead of Elege. 

Thanks!

werdeil, 2019-10-27:
I've added the 'ES' languages in the source through commit daa10a3 it will be available in the next release.

debouttek, 2019-11-02:
This is in dutch for Belgium and Netherlands 

    'nl': {
        'intro': “Neem een foto",
        'intro_print': ‘Of je kan\nnog altijd deze\nfoto printen',
        'choose': “Kies even ontwerp",
        '1': "1 foto",
        '2': "2 foto’s",
        '3': "3 foto’s",
        '4': "4 foto’s",
        'chosen': “We gaan ervoor!",
        'smile': "Lachen !",
        'processing': "Verwerken...",
        'print': "Print de foto?",
        'finished': "Bedankt",
        'oops': “Oeps er ging iets mis",
    },

werdeil, 2019-11-03:
Thanks @debouttek,  `pibooth` is now available in dutch!

robotfreak, 2019-12-03:
This is the german translation
```
   ‘de': {
        'intro': "Mach ein Foto",
        'intro_print': 'Oder drucke\ndieses Foto',
        'choose': "Wähle ein Layout",
        '1': "1 Foto",
        '2': "2 Foto’s",
        '3': "3 Foto’s",
        '4': "4 Foto’s",
        'chosen': "Los geht’s!",
        'smile': "Lächeln !",
        'processing': "Verarbeitung...",
        'print': "Foto drucken?",
        'finished': "Danke",
        'oops': "Ups, etwas ging schief",
    },
```

guldemburg, 2019-12-12:
Hi.. Hungarian language:

 'hu': {
        'intro': "Akarsz egy képet?",
        'intro_print': 'Vagy \ kinyomtathatja \ ezt a fényképet',
        'choose': "Kérlek válassz",
        '1': "1 kép",
        '2': "2 kép",
        '3': "3 kép",
        '4': "4 kép",
        'chosen': "Készülj!",
        'smile': "Csiizzz !",
        'processing': "Feldolgozás...",
        'print': "Nyomtatod a képet?",
        'finished': "Köszi",
        'oops': "Sajnos valami hiba történt",
    },

anxuae, 2019-12-13:
Thanks @guldemburg , this will be integrated in next release.

anxuae

werdeil, 2020-04-01:
Hey @guldemburg ,@debouttek, @nihuel and @Real2222 !

We have introduced a new text in pibooth in the print screen, current translations are the following (either translate by myself of google translate), can you have a look and tell me if you want any correction?
    'de': {
        'print_forget': "Bitte\nvergessen Sie\ndieses Foto",
    },
    'en': {
        'print_forget': "Please\nforget this\nphoto",
    },
    'es': {
        'print_forget': "Por favor\nolvida\nesta foto",
    },
    'fr': {
        'print_forget': "Oublie\ncette photo\ns'il te plait",
    },
    'hu': {
        'print_forget': "Kérjük,\nfelejtsd el ezt\na képet",
    },
    'nl': {
        'print_forget': "Vergeet\ndeze foto\nalstublieft",
    },

asbjornstokka, 2020-06-09:
'no': {
        'intro': "Ta et bilde",
        'intro_print': 'Eller du kan\nfremdeles printe\ndette bildet',
        'choose': "Velg ditt utseende",
        '1': "1 bilde",
        '2': "2 bilder",
        '3': "3 bilder",
        '4': "4 bilder",
        'chosen': "Start!",
        'smile': "Smil !",
        'processing': "Behandler...",
        'print': "Printe bildet?",
        'finished': "Takk",
        'oops': "Ops, noe gikk galt",
    },

werdeil, 2020-06-10:
Hi @asbjornstokka and thanks for the "no" translations! 
We have a new wording to translate: 
'print_forget': "Please\nforget this\nphoto"

Google translate gives "Glem dette bildet" is it correct or do you have better translation?

debouttek, 2020-06-10:
> Hey @guldemburg ,@debouttek, @nihuel and @Real2222 !
>
> We have introduced a new text in pibooth in the print screen, current translations are the following (either translate by myself of google [...]

Dutch is OK!

asbjornstokka, 2020-06-10:
> Hi @asbjornstokka and thanks for the "no" translations!
> We have a new wording to translate:
> 'print_forget': "Please\nforget this\nphoto"
> 
> Google translate gives "Glem dette bildet" is it correct or do you have better translation?

That works

Exitras, 2020-08-20:
Here are some changes for the german translation which I use in my PiBooth. Some are debatable but I wouldn't use such a "formal" language. Furthermore, "Foto" and "Fotos" should be upper case. 

`DEFAULT = {
    'de': {
        'intro': "Foto aufnehmen",
        'intro_print': "Dieses Foto kann mit dem rechten Knopf ausgedruckt werden",
        'choose': "Wähle dein Layout",
        '1': "1 Foto",
        '2': "2 Fotos",
        '3': "3 Fotos",
        '4': "4 Fotos",
        'chosen': "Los gehts!",
        'smile': "Bitte lächeln!",
        'processing': "Bearbeitung...",
        'print': "Foto drucken?",
        'print_forget': "Dieses Foto bitte löschen",
        'finished': "Danke",
        'oops': "Ups! Irgendwas lief schief...",
    },`

werdeil, 2020-08-20:
Thanks @Exitras for the changes, I've taken into account the uppercase and `print_forget` changes, about the `intro_print` your version talks about "right button" , as the pibooth can be used without physical buttons I'd rather avoid having them in the text, do you have another idea?

Exitras, 2020-08-20:
I agree. This makes only sense when there is a physical right button :) I would propose "Dieses Foto kann immer noch ausgedruckt werden" instead.

DJ-Dingo, 2020-09-28:
Dansk / Danish  

 'dk': {
        'intro': "Tag et foto",
        'intro_print': 'Eller du kan\nstadig udskrive\ndette foto',
        'choose': "Vælg dit layout",
        '1': "1 Foto",
        '2': "2 Fotos",
        '3': "3 Fotos",
        '4': "4 Fotos",
        'chosen': "Gør dig klar!",
        'smile': "Smil !",
        'processing': "Fremkalder Foto...",
        'print': "Print foto?",
        'print_forget': "Venligst\nglem dette\nfoto",
        'finished': "Tak",
        'oops': "Ups! Noget gik galt",
    },

werdeil, 2020-10-02:
Thanks @DJ-Dingo added 'dk' languages is added in the project !

fab1o978, 2021-11-18:
This is for the italian language!

`'it': {
    'intro': "Scatta una foto",
    'intro_print': 'Oppure puoi\nstampare\nquesta foto',
    'choose': "Scegli il formato",
    '1': "1 foto",
    '2': "2 foto",
    '3': "3 foto",
    '4': "4 foto",
    'chosen': "Andiamo!",
    'smile': "Sorridi !",
    'processing': "Elaborazione...",
    'print': "Stampo la foto?",
    'print_forget': "Per favore\nignora questa\nfoto",
    'finished': "Grazie",
    'oops': "Oops qualcosa è andato storto",
},`

klemva, 2023-02-24:
'CZ': {
        'intro': "Pojďte se vyfotit",
        'intro_print': 'Chcete foto
znovu
vytisknout?',
        'choose': "Vyberte si typ fotky",
        '1': "1 Fotka",
        '2': "2 Fotky",
        '3': "3 Fotky",
        '4': "4 Fotky",
        'chosen': "Jdeme na to!",
        'smile': "Úsměv!",
        'processing': "Pracuji na tom...",
        'print': "Chcete foto vytisknout?",
        'print_forget': "Chcete foto
zahodit?",
        'finished': "Díky",
        'oops': "Ups! Něco se stalo...",
    },

esnil, 2023-08-14:
Here comes a suggestion for translation to Swedish:

[se]
intro = Ta en bild
intro_print = Eller så kan du
    fortfarande 
    skriva ut den här bilden
choose = Välj layout
1 = 1 bild
2 = 2 bilder
3 = 3 bilder
4 = 4 bilder
chosen = Kör!
smile = Smile !
processing = Jobbar...
print = Skriv ut bilden!
print_forget = Glöm
    den här
    bilden
finished = Tack
oops = Oops, nånting blev fel