pibooth/pibooth #91: Lag with opencv

MickaelB, 2019-10-22:
First of all, i'd like to  say thank you for your previous solving of blue raw.

**opencv preview is lagging about 2 seconde**
on preview i have a lag of about 2s with logitech webcam C920

I try this script in python without lagging:
import numpy as np
import cv2

cap = cv2.VideoCapture(0)

while(True):
    # Capture frame-by-frame
    ret, frame = cap.read()

    # Display the resulting frame
    cv2.imshow('frame',frame)
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

# When everything done, release the capture
cap.release()
cv2.destroyAllWindows()

**Steps to reproduce the bug**
1. take a photo
2. preview is late about 2-3 sec
3. See error.

Is  alpha or resizing, can cause this delay?
Is it possible to downsample preview to erase latency?

**Environment**
 - OS [e.g. Raspbian Buster]
 - Camera [Logitech C920]
 - Pibooth version [1.1.4]

Thanks

anxuae, 2019-10-24:
Hi MickaelB,

You are probably right, the resizing of each frame may lag the window.
Can you give me the resolution of the picture taken with your camera (or post a picture)?

Use `cv2.imwrite('image.jpg', cv2.cvtColor(frame, cv2.COLOR_RGB2BGR))` to same a frame in an image.

Antoine

MickaelB, 2019-10-25:
Hi Antoine
I have try save image with the little script + cv2.imwrite.
It's save a 640x480...
Full image from raw pybooth are in 1934x2466

I had this 2 lines in the "basic script" to force resolution at 1920x1080
       cap.set(3,1920) #Setting webcam's image width 
       cap.set(4,1080) #Setting webcam' image height
it was lagging too..

I have try to modify opencv.py
def __init__(self,
                 iso=200,
                 resolution=(640, 480),
But it's lagging too
C920 is a 15Mp webcam
Is it possible to decrease preview resolution ans take full size image?
Any idea to solve this problem? 

Many thanks

Mickaël

MickaelB, 2019-11-02:
Hi,

I changed directly to opencv.py

self._cam.set(cv2.CAP_PROP_FRAME_WIDTH, 1280) self._cam.set(cv2.CAP_PROP_FRAME_HEIGHT, 720)

there are a little less lag.
I try to reduce with other resolution, but i have error:
OpenCV Error : Size of input arguments do not match...

Is it possible that overlay is not in same format?

thanks for your help

Mickaël

anxuae, 2019-11-05:
Hi @MickaelB 

Reducing the resolution can improve preview performances. However, the correct resolution shall be defined according to the abilities of the used camera, which can not easily guessed.

I think that the error you get is due to the chosen resolution does not fit with you camera ability.

You can try by setting first a max size (`10000` is arbitrary, your camera should automatically adapts its resolution according to its max width/height):

```python
self._cam.set(cv2.CAP_PROP_FRAME_WIDTH, 10000)
self._cam.set(cv2.CAP_PROP_FRAME_HEIGHT, 10000)
```
Then get the real max width / height:
```python
width = self._cam.get(cv2.CAP_PROP_FRAME_WIDTH)
height = self._cam.get(cv2.CAP_PROP_FRAME_HEIGHT)
```
Finally try to set a factor of this resolution:
```python
self._cam.set(cv2.CAP_PROP_FRAME_WIDTH, width / 10)
self._cam.set(cv2.CAP_PROP_FRAME_HEIGHT, height / 10)
```

anxuae

MickaelB, 2019-11-05:
Hi anxue,

I have search and discover that lag is alway in fullscreen mode or 1280x1024.

I have try your solution, preview is ok ( with horrible pixel ;-) ) in windows (800x600) but when i pass in fullscreen it's also slow.

i'm in debug mode, and the number of buffer images is decrease by 2 or 3 when fullscreen is enable.

i see "Add to buffer the image 'XXXXXX' " (FPS)  decrease between this two modes.

how the number of images in the buffer is calculate?
It seems to be that refresh is slower in fullscreen, or it 's because of second resizing to be in fullscreen?

Many thanks
Mickaël

alexszilagyi, 2020-01-24:
@MickaelB : Does it needs more Video card memory? Is there any workaround for adding any other external device?

MickaelB, 2020-01-26:
@alexszilagyi  : you can try to disable countdown,fullscreen and decrease resolution to reduce lag.

alexszilagyi, 2020-01-26:
@MickaelB : I see but I was wondering if I could somehow overclock the Raspberry Pi to reach the desired performance?

Did a test with a PiCamera with the same config and I don't have any lag.

I'm not sure how to overcome this issue, I need full screen & countdown, as for resolution will tweak and toggle until I reach to a resonable point.

Thanks for your great project! 👍

MickaelB, 2020-01-31:
@alexszilagyi 

I have reuse a dirty python script like this without lag problem (work in progress):
<script src="https://gist.github.com/MickaelB/a3e9bc37f8c11705204072c5ba60f91a.js"></script>
from this repo
https://github.com/bdelphin/RPi-PhotoBooth

werdeil, 2020-01-31:
Thanks for the link to this repo, it show a new reflexion track. I'll try to have a look on it.

Keep us update if you manage to have something we could integrate.

thibault-78, 2020-05-25:
> @alexszilagyi
> 
> I have reuse a dirty python script like this without lag problem (work in progress):
> 
> <script src="https://gist.github.com/MickaelB/a3e9bc37f8c11705204072c5ba60f91a.js"></script>
> from this repo
> https://github.com/bdelphin/RPi-PhotoBooth

I experienced the same behavior.
Any code to share solving the issue?

alexszilagyi, 2020-05-25:
@thibault-78 : Not for now. Looking forward for a possible solution. Will add a PR if I have something that can help.

anxuae, 2020-11-14:
@MickaelB,

A fix has been done to solve the #186 issue. Maybe you can try if it solves also your issue?
(to test it, please install a developing version)

anxuae, 2021-05-07:
OpenCV camera improved in `pibooth` v2.0.2

Please re-open the issue if you are still facing to the same issue.

BR
anxuae

closed 2021-05-07 by anxuae

JFK344, 2022-02-08:
In my case the camera was not using the right codec. 
I used this command to get all the available Codecs.

`v4l2-ctl --list-formats-ext`

my camera used YUYV which was capable of 6 fps at 1920x1080.

Changed the settings to `cam.set(cv2.CAP_PROP_FOURCC, cv2.VideoWriter.fourcc('M','J','P','G'))` whichs had NO result eighter. 

insted of accesssing the camer by index i used `reversingCam = cv2.VideoCapture("/dev/video0")` which made things working. 

heres my whole Code:

```python
import cv2
  
reversingCam = cv2.VideoCapture("/dev/video0")
reversingCam.set(cv2.CAP_PROP_FOURCC, cv2.VideoWriter.fourcc('M','J','P','G'))
reversingCam.set(3, 1280)
reversingCam.set(4, 720)

reversingCam.set(cv2.CAP_PROP_BUFFERSIZE, 1)

while(True):
    ret, frame = reversingCam.read()
    
    cv2.imshow('frame', frame)

    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

reversingCam.release()
cv2.destroyAllWindows()
```