pibooth/pibooth #92: processing improvement

MickaelB, 2019-10-22:
**processing improvement.**
it could be cool to have the same number of picture in processing background and number of taken picture.
Generate thumbnail and show in the processing screen

**Describe the solution you'd like**
perhaps we can make different background and call them.

it's just to be perfect. but it can be difficult with raspi 

Many thanks for your hard work on pibooth

werdeil, 2019-10-27:
> the same number of picture

Not sure this would work with only one picture

> Generate thumbnail and show in the processing screen

My main worry would be that the Raspberry Pi is already working a lot at this screen, making is work on another thing that generating the picture would increase the processing time.

I have another idea though, maybe we could put in the processing screen the image generated previously, as they are already generated it may be possible to add them at a lower cost. What do you think @anxuae ?

anxuae, 2019-10-30:
It is possible to generate Thumbnail using PIL:

```
image = Image.open(data)
image.thumbnail((36,36), Image.NEAREST)
```
However, we designed `pibooth` to be an embedded software in  the raspberry-pi. Some tests could be done to ensure that the processing period is not too much extended. But my guess is that current version of raspberry pi is not enough powerful.

MickaelB, 2019-11-02:
ok, perhaps we can generate thumbnail in background during waiting from main screen ?

closed 2019-12-01 by MickaelB