pibooth/pibooth #93: [question] pycups documentation

tanuj101, 2019-10-23:
Hi! I wanted to know what is your source of documentation for pycups lib mine is [this](http://nagyak.eastron.hu/doc/system-config-printer-libs-1.2.4/pycups-1.9.51/html/cups.Connection-class.html#createSubscription) 
Just curious.

tanuj101, 2019-10-23:
not a real issue though, I couldn't see any contact info

anxuae, 2019-10-23:
Hi tanuj101, 

The only documents used are:
- nagyak.eastron.hu/doc/system-config-printer-libs-1.2.4/pycups-1.9.51/html
- http://blog.whatgeek.com.pt/2014/07/linux-cups-notifications
- https://www.cups.org/doc/spec-design.html
- https://www.uucp.hu/cups/help/ref-subscriptions-conf.html

And lots of time to test and re-test different parameters/code architecture as the documentation is not clear at all.

anxuae

tanuj101, 2019-10-23:
Thanks for the quick response. I actually wondered right into your repo searching about subscriptions tbh. Yes I agree the documentation is not clear at all maybe because it is something that is not used widely by people.

anxuae, 2019-10-27:
Hi tanuj101,

I have extract the notification system in a dedicated repository : https://github.com/anxuae/pycups-notifier.

I propose to close this issue.

tanuj101, 2019-10-28:
Oh yeah as I've mentioned earlier it wasn't an issue you could've closed it anytime though thanks for the repo :)

closed 2019-10-28 by tanuj101