pibooth/pibooth #94: Disable ISO Setting

DEhrhardt, 2019-10-31:
Hi,

Is it possible to disable the ISO Setting in the pibooth config? 
I‘‘d prefer to configure all the settings on my camera in manual mode.

I‘d be happy about a fast reply, because I want to use the pibooth on the first event tomorrow.

anxuae, 2019-11-01:
Hi DEhrhardt,

You can try by setting an invalid value, for instance `-1`.
Otherwise, If you are starting `pibooth` from the cloned repository, you can comment the lines responsible for the settings.

Which kind of camera are  you using? (please use issue template to open a bug, it avoids to forget information in the description)

DEhrhardt, 2019-11-01:
Sorry for the issue ... I'm fine. Somehow the way to adjust the iso on the camera changed. I can change it as I need in manual mode. Issue can be closed. I'm excited to use the pibooth for the first time this evening.

anxuae, 2019-11-01:
No problem, have a nice day with `pibooth` ;-)

closed 2019-11-01 by anxuae

Exitras, 2020-08-20:
I experienced the same issue. Even though I set iso in `pibooth --config` to `-1` and in the camera to auto, it sticks to the last value that has been in `pibooth -- config` (in my case iso 800). I am using a Canon EOS 2000D. It is fine as it is but would be slightly better if I could set iso to automatic mode (the rest of the camera is in manual mode).
Do you know which line to comment in order to get set it to automatic iso?

werdeil, 2020-08-20:
HI @Exitras, putting an invalid value should work, as `pibooth` will try to set it anyways and will get an error from gPhoto (but shall not be an issue).

The line to comment would be the follwing one in the gphoto.py file:
`self.set_config_value('imgsettings', 'iso', self._iso)`
 but I would rather find out why this is not working.