pibooth/pibooth #95: How to change background layout?

debouttek, 2019-11-02:
A question.
How do I change the background of the photo print? 
For example: When I would like to use pibooth_landscape.jpg for printing?

[image]

anxuae, 2019-11-03:
Hi @debouttek 

As explained here https://github.com/werdeil/pibooth#final-picture-rendering, the rendering of the final picture is dependent of 2 parameters in the config.

Landscape will be if you set `[PICTURE][orientation]` or if you chose a landscape resolution in `[CAMERA][resolution]` (width bigger than height).

Remember that you can edit all config parameters with `pibooth --config`

anxuae

debouttek, 2019-11-03:
> Hi @debouttek
>
> As explained here https://github.com/werdeil/pibooth#final-picture-rendering, the rendering of the final picture is dependent of 2 parameters in the config.
>
> Landscape will be if you [...]

What about the backgrounds for the printing. Is it called overlays? How to add them to the print?

werdeil, 2019-11-03:
You have two options in the config:
- Change the background with the option [PICTURE][background] where you can put either a color in RGB (e.g. (255, 0, 0) for red) or a path to a background image
- Change the overlay that will be put on the photos with the option [PICTURE][overlay] where you can use the same kind of values as in background.

On the example pibooth_lanscape.jpg, there was no overlay and the background option was set to white (255, 255, 255).

anxuae, 2019-11-05:
Hi @debouttek,

Can we close this issue?

Thanks

closed 2019-11-05 by debouttek